tapparelj/gr-lora_sdr #8: Compilation error

DavidKrepzi, 2021-09-25:
Hi, I am using Arch Linux with all the dependencies installed.  However I do receive the following error message when I start "make"

```
make
Consolidate compiler generated dependencies of target gnuradio-lora_sdr
[  3%] Building CXX object lib/CMakeFiles/gnuradio-lora_sdr.dir/add_crc_impl.cc.o
/home/user/gr-lora_sdr/lib/add_crc_impl.cc: In static member function ‘static gr::lora_sdr::add_crc::sptr gr::lora_sdr::add_crc::make(bool)’:
/home/user/gr-lora_sdr/lib/add_crc_impl.cc:16:46: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::lora_sdr::add_crc_impl]()’ from ‘std::shared_ptr<gr::lora_sdr::add_crc_impl>’ to ‘gr::lora_sdr::add_crc::sptr’ {aka ‘boost::shared_ptr<gr::lora_sdr::add_crc>’}
   16 |             return gnuradio::get_initial_sptr(new add_crc_impl(has_crc));
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                              |
      |                                              std::shared_ptr<gr::lora_sdr::add_crc_impl>
make[2]: *** [lib/CMakeFiles/gnuradio-lora_sdr.dir/build.make:76: lib/CMakeFiles/gnuradio-lora_sdr.dir/add_crc_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:233: lib/CMakeFiles/gnuradio-lora_sdr.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

Would this still be a dependency issue? Thanks for providing some hints to solve this problem. 

```

tapparelj, 2021-09-25:
Hi,

I would suspect an issue with the version of either boost or gcc. The file environment.yml provide the full description of the conda environment used. You can first try to use the matching versions for libgcc, libgcc-devel, gcc_linux-64, gcc_impl_linux-64, gxx_linux-64, gxx_impl_linux-64, boost and boost-cpp. Another option is to duplicate this conda environment and use it to build and use this module. 

Let me know if having the matching versions of the library fixed the building issue, or if you have difficulties with the duplication of the environment as I have never used this module on Arch Linux.

DavidKrepzi, 2021-09-29:
Thank you for your prompt response. Yes, I think you are right that I have a software version mismatch. My best bet is to try Conda. I will give it a go. Again, thank you for making available this very useful tool for the LoRa research community.

Lijinyudoudou, 2021-11-19:
hello，Thank you very much for this file and it real helps tp learn about gnuradio.However, when i tried to run  make i meet this problem：

CMake Error at python/cmake_install.cmake:48 (file):
  file INSTALL cannot find
  "/home/LEELA/desktop/gr-lora_sdr/build/python/__init__.pyc": No such file
  or directory.
Call Stack (most recent call first):
  cmake_install.cmake:51 

What kind of problem it is? I run it with VMware with Ubuntu 20.04.3

tapparelj, 2021-11-22:
Hello @Lijinyudoudou,
did you received any warning/errors when executing the `cmake ..` command in your build folder? Did you build gnuradio using pybombs, conda, from source, or with a different method?

Lijinyudoudou, 2021-11-25:
> 

hello， when cmake .. there is no warning or errors , and I install gnuradio with ppa installation.

tapparelj, 2021-11-26:
@Lijinyudoudou can you provide the output of the `cmake ..` command as well as the output of the `make`  command after deleting the content of the folder <path_to_repository>/gr-lora_sdr/build/? 
If you can also provide the file python/cmake_install.cmake that is causing the error, that would be helpful.

Lijinyudoudou, 2021-11-26:
> @Lijinyudoudou can you provide the output of the `cmake ..` command as well as the output of the `make` command after deleting the content of the folder <path_to_repository>/gr-lora_sdr/build/? If [...]

Lijinyudoudou, 2021-11-26:
@tapparelj  Hello, I tried for much more time and tried to install something but now there are errors after cmake .. 
<pre>-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version &quot;0.29.1&quot;) 
-- Checking for module &apos;gmp&apos;
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version &quot;1.71.0&quot;, minimum required is &quot;1.71.0&quot;) found components: date_time program_options filesystem system regex thread unit_test_framework 
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version &quot;3.8.10&quot;) 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version &quot;3.8.10&quot;) 
-- Checking for module &apos;thrift&apos;
--   Found thrift, version 0.13.0
-- Python checking for Thrift - found
-- Found THRIFT: /usr/lib/x86_64-linux-gnu/libthrift.so  
-- Checking for module &apos;fftw3f &gt;= 3.0&apos;
--   Found fftw3f , version 3.3.8
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Using install prefix: /usr/local
-- Building for version: ca195cb1 / 1.0.0git
-- No C++ unit tests... skipping
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Found SWIG: /usr/bin/swig4.0  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version &quot;3.8.10&quot;) 
  File &quot;&lt;string&gt;&quot;, line 2
SyntaxError: bytes can only contain ASCII literal characters.
  File &quot;&lt;string&gt;&quot;, line 2
SyntaxError: bytes can only contain ASCII literal characters.
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPython.cmake:142 (add_custom_target):
  add_custom_target cannot create target &quot;ALL&quot; because another target with
  the same name already exists.  The existing target is a custom target
  created in source directory &quot;/home/jinyuli/桌面/gr-lora_sdr/swig&quot;.  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPython.cmake:329 (GR_UNIQUE_TARGET)
  python/CMakeLists.txt:31 (GR_PYTHON_INSTALL)

-- Configuring incomplete, errors occurred!
See also &quot;/home/jinyuli/桌面/gr-lora_sdr/build/CMakeFiles/CMakeOutput.log&quot;.
See also &quot;/home/jinyuli/桌面/gr-lora_sdr/build/CMakeFiles/CMakeError.log&quot;.
</pre>

And following are the Error files mentoned above , the output after the cmake ..

[CMakeError.log](https://github.com/tapparelj/gr-lora_sdr/files/7608442/CMakeError.log)
[CMakeOutput.log](https://github.com/tapparelj/gr-lora_sdr/files/7608445/CMakeOutput.log)

tapparelj, 2021-11-26:
It now seems to have issue with the characters '桌面' in your installation path.

Lijinyudoudou, 2021-11-26:
> 
  so Should I  install ubuntu in english? This characters just desktop in Chinese..

tapparelj, 2021-11-26:
Or maybe installing the module in a path that doesn't includes Chinese characters may fix it.

Lijinyudoudou, 2021-11-26:
> Or maybe installing the module in a path that doesn't includes Chinese characters may fix it.
@tapparelj  
Thank you very much. I have changed the language to english and tried again with git .
I have to install liborc-0.4-dev and then later i have installed this Module successfully.
<pre>sudo apt-get install liborc-0.4-dev
</pre>

Thank you for your time and answer. It really helps.

tapparelj, 2021-11-26:
@Lijinyudoudou Nice to hear, let me know if you run into any other issues.

closed 2021-11-26 by tapparelj