tapparelj/gr-lora_sdr #9: ImportError: libgnuradio-lora_sdr.so.1.0.0git: cannot open shared object file: No such file or directory

aviadef, 2021-10-05:
Hi,
Issue traceback below happens after successfully generating tx_rx_simulation.grc to tx_rx_simultaion.py and execution on gnu-radio companion.

Already solved  "ImportError: No module named lora_sdr" - issue.
Using Ubuntu 20.04.3 LTS.
Python 3
Gnu-radio 3.8.1.0
 

```

Traceback (most recent call last):
  File "./tx_rx_simulation.py", line 22, in <module>
    import lora_sdr
  File "/root/gr-lora_sdr/lib/python3/dist-packages/lora_sdr/__init__.py", line 34, in <module>
    from .mu_demod import mu_demod
  File "/root/gr-lora_sdr/lib/python3/dist-packages/lora_sdr/mu_demod.py", line 6, in <module>
    from .lora_sdr_swig import *
  File "/root/gr-lora_sdr/lib/python3/dist-packages/lora_sdr/lora_sdr_swig.py", line 13, in <module>
    from . import _lora_sdr_swig
ImportError: libgnuradio-lora_sdr.so.1.0.0git: cannot open shared object file: No such file or directory

```

martynvdijke, 2021-10-05:
Hi,

Could you do `ls -la` inside `/root/gr-lora_sdr/lib/python3/dist-packages/lora_sdr/` to show the directory contents of the python package.
This is related to the swig and python interface, python is not able to find the swig generated libary.

aviadef, 2021-10-05:
Hi,

drwxr-xr-x 1 root root     512 Oct  4 15:48 .
drwxr-xr-x 1 root root     512 Sep 22 09:44 ..
-rw-r--r-- 1 root root    1212 Aug 10 10:24 __init__.py
-rw-r--r-- 1 root root     365 Oct  4 15:48 __init__.pyc
-rw-r--r-- 1 root root     365 Oct  4 15:48 __init__.pyo
drwxr-xr-x 1 root root     512 Oct  4 16:00 __pycache__
-rw-r--r-- 1 root root 2500504 Oct  4 15:48 _lora_sdr_swig.so
-rw-r--r-- 1 root root  351990 Oct  4 15:47 lora_sdr_swig.py
-rw-r--r-- 1 root root  412550 Oct  4 15:48 lora_sdr_swig.pyc
-rw-r--r-- 1 root root  412550 Oct  4 15:48 lora_sdr_swig.pyo
-rw-r--r-- 1 root root    9818 Aug 10 10:24 mu_demod.py
-rw-r--r-- 1 root root    7501 Oct  4 15:48 mu_demod.pyc
-rw-r--r-- 1 root root    7501 Oct  4 15:48 mu_demod.pyo

mutescumarian, 2021-11-03:
@aviadef Hi! Can you elaborate on how you solved the import error problem? Thank you!

tapparelj, 2021-11-12:
Hi @mutescumarian , 
if your issue concerns "ImportError: No module named lora_sdr", this might help: When building the module, cmake will install the library in the active conda environment or PyBombs prefix if possible. With conda, the paths should be set automatically when activating the environment. However, if the path are still missing you have to set both the variables PYTHONPATH and LD_LIBRARY_PATH as described here https://wiki.gnuradio.org/index.php/ModuleNotFoundError. If you want to install the library in a specific path, you can set it with the parameter -DCMAKE_INSTALL_PREFIX when executing cmake.

Lijinyudoudou, 2021-12-02:
@tapparelj Hello , I meet the problem you mentioned above.And I tried to follow the steps and then reboot the computer,but then it doesn't work. I am comfused about the store the commands  in a Bash. Perhaps I did them wrong.
<pre><font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ gnuradio-companion
&lt;&lt;&lt; Welcome to GNU Radio Companion 3.8.3.1 &gt;&gt;&gt;

Block paths:
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: &quot;/home/jinyuli/Desktop/gr-lora_sdr/apps/single_user/lora_TX.grc&quot;
&gt;&gt;&gt; Done

Generating: &apos;/home/jinyuli/Desktop/gr-lora_sdr/apps/single_user/lora_TX.py&apos;

Executing: /usr/bin/python3 -u /home/jinyuli/Desktop/gr-lora_sdr/apps/single_user/lora_TX.py

Traceback (most recent call last):
  File &quot;/home/jinyuli/Desktop/gr-lora_sdr/apps/single_user/lora_TX.py&quot;, line 22, in &lt;module&gt;
    import lora_sdr
ModuleNotFoundError: No module named &apos;lora_sdr&apos;

&gt;&gt;&gt; Done (return code 1)
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ gnuradio-config-info --prefix
/usr
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ find /usr -name gnuradio | grep &quot;packages&quot;
/usr/lib/python3/dist-<font color="#CC0000"><b>packages</b></font>/gnuradio
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ export LD_LIBRARY_PATH=/usr/lib:$LD_LIBRARY_PATH
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ export PYTHONPATH=/usr/lib/python3/dist_packages:/usr/lib/python3/dist-packages:$PYTHONPATH
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ export LD_LIBRARY_PATH=/usr/lib:$LD_LIBRARY_PATH
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ 
</pre>
And then exit,,I am not sure how to save these two path correctly.
<pre><font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ ~/.profile
bash: /home/jinyuli/.profile: Permission denied
</pre>

tapparelj, 2021-12-02:
Hello @Lijinyudoudou ,
A few notes:
- You should export the paths before opening gnuradio-companion in the same terminal.
- in the paths you type, verify that there is the folder lora_sdr in the PYTHONPATH you set and the file libgnuradio-lora_sdr.so in LD_LIBRARY_PATH. Those are the files/folder you need to provide the path to.
- You can append the export commands in the file ~/.bashrc (you can create it if needed), that way each time you will launch a new terminal, those commands will be executed. (you can use echo $PYTHONPATH to check the current paths.)
- When you executed the command `cmake ..` did you specified a prefix path?

Lijinyudoudou, 2021-12-02:
> 

@tapparelj Hello when executing the make .. I didn't specfied a prefix path

Lijinyudoudou, 2021-12-02:
> > Hello @Lijinyudoudou , A few notes:
> > 
> > * You should export the paths before opening gnuradio-companion in the same terminal.
> > * in the paths you type, verify that there is the folder lora_sdr in the PYTHONPATH you set and the file libgnuradio-lora_sdr.so in LD_LIBRARY_PATH. Those are the files/folder you need to provide the path to.
> > * You can append the export commands in the file ~/.bashrc (you can create it if needed), that way each time you will launch a new terminal, those commands will be executed. (you can use echo $PYTHONPATH to check the current paths.)
> > * When you executed the command `cmake ..` did you specified a prefix path?
> >   Hello, I have used
> 
>  ls profile
> profile
> jinyuli@jinyuli-virtual-machine:~$ ls -a
> .              .config    .gr_fftw_wisdom       Pictures                   Templates
> ..             Desktop    .gr_fftw_wisdom.lock  .profile                   Videos
> .bash_history  Documents  .local                profile
> .bash_logout   Downloads  .mozilla              Public
> .bashrc        .gnupg     Music                 .ssh
> .cache         .gnuradio  .pam_environment      .sudo_as_admin_successful
> jinyuli@jinyuli-virtual-machine:~$ vi profile
> jinyuli@jinyuli-virtual-machine:~$ 
> to store the path and also reboot the comouter, but this problem still remain

and this are the path i used.
<pre>export LD_LIBRARY_PATH=/usr/lib:$LD_LIBRARY_PATH
export PYTHONPATH=/usr/lib/python3/dist_packages:/usr/lib/python3/dist-packages:$PYTHONPATH
<font color="#729FCF">~                                                                                                 </font>
<font color="#729FCF">~         </font></pre>

Lijinyudoudou, 2021-12-02:
> 

> Hello @Lijinyudoudou , A few notes:
>
> * You should export the paths before opening gnuradio-companion in the same terminal.
> * in the paths you type, verify that there is the folder lora_sdr in the [...]

And I checked the path have been changed through following command
<pre>&gt;&gt;&gt; Done (return code 1)
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ echo $PYTHONPATH
/usr/lib/python3/dist_packages:/usr/lib/python3/dist-packages:
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ echo $LD_LIBRARY_PATH
/usr/lib:
<font color="#4E9A06"><b>jinyuli@jinyuli-virtual-machine</b></font>:<font color="#3465A4"><b>~/Desktop</b></font>$ 
</pre>

Lijinyudoudou, 2021-12-03:
@tapparelj 
Hello, I think the path I have changed correctly but i met a new problem never seem before,

no module named "lora_sdr.lora_sdr.swig"
if it is also some path i didn't set correctly?

Lijinyudoudou, 2021-12-03:
> @tapparelj Hello, I think the path I have changed correctly but i met a new problem never seem before,
> 
> no module named "lora_sdr.lora_sdr.swig" if it is also some path i didn't set correctly?

[image]

MohamedAMohd, 2023-10-06:
> Hi, Issue traceback below happens after successfully generating tx_rx_simulation.grc to tx_rx_simultaion.py and execution on gnu-radio companion.
>
> Already solved "ImportError: No module named [...]

I need your help @aviadef how did you solve "Traceback (most recent call last):
  File "/home/lora/Desktop/gr-lora_sdr/examples/Hardware.py", line 39, in <module>
    import gnuradio.lora_sdr as lora_sdr
ModuleNotFoundError: No module named 'gnuradio.lora_sdr'"

tapparelj, 2023-10-06:
Hello @MohamedAMohd,
first a note: since GNURadio 3.9, swig has been replaced with pybind. The code of this repository is only updated for gnuradio 3.10, therefore, you probably have a traceback different from the one you cited.
Can you give more details about how you installed gnuradio, which version you installed, and on which OS?
Did you already tried what is explained here [https://wiki.gnuradio.org/index.php/ModuleNotFoundError]() ?

MohamedAMohd, 2023-10-06:
> Hello @MohamedAMohd, first a note: since GNURadio 3.9, swig has been replaced with pybind. The code of this repository is only updated for gnuradio 3.10, therefore, you probably have a traceback [...]

@tapparelj I have installed gnuradio version 3.10 in Ubuntu desktop 20.04 with help of this [https://github.com/tapparelj/gr-lora_sdr](url)

smsajjadi, 2024-11-10:
Hi. 
I have encountered a similar issue. I am trying to install the code on ubuntu 24.04, where my python is 3.12.3 and gnuradio is 3.10.9.2. So I decided to bypass conda and compile and install on my /usr/local folder. But what i get is:

`python3 examples/tx_rx_functionality_check.py 
Traceback (most recent call last):
  File  "{...}/examples/tx_rx_functionality_check.py", line 23, in <module>
    import gnuradio.lora_sdr as lora_sdr
  File "/usr/local/lib/python3.12/dist-packages/gnuradio/lora_sdr/__init__.py", line 18, in <module>
    from .lora_sdr_python import *
ImportError: libgnuradio-runtime.so.3.10.3: cannot open shared object file: No such file or directory
`

Confusing for me is that why is it looking for a 3.10.3 version of libgnuradio-runtime? I find no mention of gnuradio version in the code except in some comments.