tapparelj/gr-lora_sdr #10: with SF=11 frame sync requires more input data than provided

edersonrmachado, 2021-11-01:
Hi, thanks for make available this great project! I've the following GNURadio message when using spreading factor SF=11:

```
sched: <block frame_sync (10)> is requesting more input data
  than we can provide.
  ninput_items_required = 8200
  max_possible_items_available = 8191
  If this is a filter, consider reducing the number of taps.
```  
It seems that `Frame_sync` block need more samples but it is not possible to provide. Do you have any idea how could I fix this issue?

tapparelj, 2021-11-02:
Hi, this issue arise because the block preceding frame_sync doesn't have enough space allocated for its output buffer. 
You can solve this by increasing the size of this buffer by setting the variable "Minoutbuf". You can specify it in gnuradio-companion by double-clicking on the block preceding frame_sync under the "Advanced" tab, or using the function "{name_of_the_block}.set_min_output_buffer(2**sf*5)" in the python script you after instantiating the block.

edersonrmachado, 2021-11-02:
Nice, thanks for your answer and explanation!

closed 2021-11-09 by tapparelj