tapparelj/gr-lora_sdr #13: gr_lora-sdr Building OOT modules to use with conda-installed GNU Radio

ewwll, 2022-03-08:
The following error occurs when constructing GR-LORA_SDR using Windows VS2019 including MSVC V141
[1/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\crc_verif_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/crc_verif_impl.cc.obj
C:\PROGRA2\MICROS4\2019\COMMUN1\VC\Tools\MSVC\14161.270\bin\HostX64\x64\cl.exe /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\crc_verif_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\crc_verif_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
D:\360\lora\gr-lora_sdr-master\lib..\include\lora_sdr/utilities.h(9): fatal error C1083: 无法打开包括文件: “sys/resource.h”: No such file or directory
[2/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\gray_enc_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/gray_enc_impl.cc.obj
C:\PROGRA2\MICROS4\2019\COMMUN1\VC\Tools\MSVC\14161.270\bin\HostX64\x64\cl.exe /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\gray_enc_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\gray_enc_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
D:\360\lora\gr-lora_sdr-master\lib\gray_enc_impl.cc(15): error C2440: “return”: 无法从“std::shared_ptrgr::lora_sdr::gray_enc_impl”转换为“boost::shared_ptrgr::lora_sdr::gray_enc”
D:\360\lora\gr-lora_sdr-master\lib\gray_enc_impl.cc(15): note: 无构造函数可以接受源类型，或构造函数重载决策不明确
[3/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\header_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/header_impl.cc.obj
C:\PROGRA2\MICROS4\2019\COMMUN1\VC\Tools\MSVC\14161.270\bin\HostX64\x64\cl.exe /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\header_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
d:\360\lora\gr-lora_sdr-master\lib\header_impl.h(16): error C3646: “m_cnt_nibbles”: 未知重写说明符
d:\360\lora\gr-lora_sdr-master\lib\header_impl.h(16): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
d:\360\lora\gr-lora_sdr-master\lib\header_impl.h(17): error C3646: “m_cnt_header_nibbles”: 未知重写说明符
d:\360\lora\gr-lora_sdr-master\lib\header_impl.h(17): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(16): error C2440: “return”: 无法从“std::shared_ptrgr::lora_sdr::header_impl”转换为“boost::shared_ptrgr::lora_sdr::header”
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(16): note: 无构造函数可以接受源类型，或构造函数重载决策不明确
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(35): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(82): error C2065: “m_cnt_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(89): error C2065: “m_cnt_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(92): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(100): warning C4805: “|”: 在操作中将类型“int”与类型“bool”混合不 安全
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(110): warning C4805: “|”: 在操作中将类型“int”与类型“bool”混合不 安全
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(119): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(121): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(122): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(132): error C2065: “m_cnt_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(140): error C2065: “m_cnt_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(141): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
[4/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\frame_sync_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/frame_sync_impl.cc.obj
C:\PROGRA2\MICROS4\2019\COMMUN1\VC\Tools\MSVC\14161.270\bin\HostX64\x64\cl.exe /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\frame_sync_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\frame_sync_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
D:\360\lora\gr-lora_sdr-master\lib..\include\lora_sdr/utilities.h(9): fatal error C1083: 无法打开包括文件: “sys/resource.h”: No such file or directory
[5/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\header_decoder_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/header_decoder_impl.cc.obj
C:\PROGRA2\MICROS4\2019\COMMUN1\VC\Tools\MSVC\14161.270\bin\HostX64\x64\cl.exe /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\header_decoder_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\header_decoder_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
[... cut ...] \include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\add_crc_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\add_crc_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
D:\360\lora\gr-lora_sdr-master\lib..\include\lora_sdr/utilities.h(9): fatal error C1083: 无法打开包括文件: “sys/resource.h”: No such file or directory
ninja: build stopped: subcommand failed.

tapparelj, 2022-03-09:
Hello,
The issue probably comes from trying to build the module on Windows instead of Linux.
According to [stackoverflow.com/a/18671155/5666087](https://stackoverflow.com/a/18671155/5666087), sys/resource.h is a POSIX header file and will not be found in Windows, unless you use Cygwin.

ewwll, 2022-03-14:
I compiled this project using Cmake in the WSL-Ubuntu environment and now the following problem occurs, all the dependent libraries have been installed
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Conda installed GNU Radio. Setting CMAKE_INSTALL_PREFIX to /root/anaconda3/envs/gnuradio
-- Build type not specified: defaulting to release.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'gmp'
--   Found gmp, version 6.2.1
-- Found GMP: /usr/local/lib/libgmpxx.so
-- Using GMP.
-- Found MPLIB: /usr/local/lib/libgmpxx.so
-- Found Boost: /root/anaconda3/envs/gnuradio/lib/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options system regex thread
-- Found Volk: Volk::volk
-- User set python executable /root/anaconda3/envs/gnuradio/bin/python
-- Found PythonInterp: /root/anaconda3/envs/gnuradio/bin/python (found version "3.8.12")
-- Found PythonLibs: /root/anaconda3/envs/gnuradio/lib/libpython3.8.so (found suitable exact version "3.8.12")
-- Checking for module 'sndfile'
--   Found sndfile, version 1.0.28
-- Found SNDFILE: /usr/local/lib/libsndfile.so
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Found Git: /usr/bin/git
-- Extracting version information from git describe...
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
-- Using install prefix: /root/anaconda3/envs/gnuradio
-- Building for version: ca195cb1 / 1.0.0git
-- No C++ unit tests... skipping
-- Found SWIG: /usr/bin/swig4.0 (found version "4.0.1")
-- Found PythonLibs: /root/anaconda3/envs/gnuradio/lib/libpython3.8.so (found version "3.8.12")
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of size_t
-- Check size of size_t - done
-- Check size of unsigned int
-- Check size of unsigned int - done
-- Found PythonLibs: /root/anaconda3/envs/gnuradio/lib/libpython3.8.so (found suitable version "3.8.12", minimum required is "2")
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
Traceback (most recent call last):
  File "<string>", line 2, in <module>
TypeError: Unicode-objects must be encoded before hashing
CMake Error at cmake/Modules/UseSWIG.cmake:199 (add_executable):
  The target name "/home/ewwll/gr-lora_sdr/build/swig/.cpp" is reserved or
  not valid for certain CMake features, such as generator expressions, and
  may result in undefined behavior.
Call Stack (most recent call first):
  cmake/Modules/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
  cmake/Modules/GrSwig.cmake:173 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:49 (GR_SWIG_MAKE)

CMake Error at cmake/Modules/GrSwig.cmake:211 (GR_LIBTOOL):
  Unknown CMake command "GR_LIBTOOL".
Call Stack (most recent call first):
  swig/CMakeLists.txt:54 (GR_SWIG_INSTALL)

-- Configuring incomplete, errors occurred!
See also "/home/ewwll/gr-lora_sdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/ewwll/gr-lora_sdr/build/CMakeFiles/CMakeError.log".

tapparelj, 2022-03-14:
` File "", line 2, in TypeError: Unicode-objects must be encoded before hashing
CMake Error at cmake/Modules/UseSWIG.cmake:199 (add_executable):
The target name "/home/ewwll/gr-lora_sdr/build/swig/.cpp" is reserved `

The fact that a file with no name is referenced seems strange. Someone also had issues when using chinese characters in their  paths #8, and fixed it by installing ubuntu in english.

Maybe the cmake logs can provide more information.

SainathVam5hi, 2022-04-13:
tscgs@ubuntu:~/gr-lora_sdr/build$ cmake ..
-- Build type not specified: defaulting to release.
-- Using GMP.
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
-- Extracting version information from git describe...
-- Using install prefix: /usr/local
-- Building for version: ca195cb1 / 1.0.0git
-- No C++ unit tests... skipping
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable version "3.8.10", minimum required is "2") 
Traceback (most recent call last):
  File "<string>", line 2, in <module>
TypeError: Unicode-objects must be encoded before hashing
CMake Error at cmake/Modules/UseSWIG.cmake:199 (add_executable):
  The target name "/home/tscgs/gr-lora_sdr/build/swig/.cpp" is reserved or
  not valid for certain CMake features, such as generator expressions, and
  may result in undefined behavior.
Call Stack (most recent call first):
  cmake/Modules/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
  cmake/Modules/GrSwig.cmake:173 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:49 (GR_SWIG_MAKE)

CMake Error at cmake/Modules/GrSwig.cmake:211 (GR_LIBTOOL):
  Unknown CMake command "GR_LIBTOOL".
Call Stack (most recent call first):
  swig/CMakeLists.txt:54 (GR_SWIG_INSTALL)

-- Configuring incomplete, errors occurred!
See also "/home/tscgs/gr-lora_sdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/tscgs/gr-lora_sdr/build/CMakeFiles/CMakeError.log".

I'm working on Ubuntu and still facing this, any help?

tapparelj, 2022-04-13:
Hello  @SainathVam5hi ,
which version of cmake and swig are you using? 
If that is not already the case can you try using cmake 3.19.7 and swig 3.0.12?

SainathVam5hi, 2022-04-13:
I used the latest versions at first and then switched to the versions you've asked me but with no luck tried on both GNU3.8 as well as GNU3.10

tapparelj, 2022-04-13:
Based on that line: TypeError: Unicode-objects must be encoded before hashing
Can you try changing the line 177 of {repo path}/gr-lora_sdr/cmake/Modules/UseSWIG.cmake
from
 `execute_process(
    COMMAND ${PYTHON_EXECUTABLE} -c "import re, hashlib
unique = hashlib.md5('${reldir}${ARGN}').hexdigest()[:5]
print(re.sub('\\W', '_', '${name} ${reldir} ' + unique))"
    OUTPUT_VARIABLE _target OUTPUT_STRIP_TRAILING_WHITESPACE
  )`
to

`execute_process(
    COMMAND ${PYTHON_EXECUTABLE} -c "import re, hashlib
unique = hashlib.md5('${reldir}${ARGN}'.encode('utf-8')).hexdigest()[:5]
print(re.sub('\\W', '_', '${name} ${reldir} ' + unique))"
    OUTPUT_VARIABLE _target OUTPUT_STRIP_TRAILING_WHITESPACE
  )`
i

SainathVam5hi, 2022-04-13:
Tried that well FYI I'm on GNU 3.9.5.0 (Python 3.8.10), swig 4.0.1, and cmake 3.19.7
---------------------------
OUTPUT
-- Build type not specified: defaulting to release.
-- Using GMP.
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
-- Extracting version information from git describe...
-- Using install prefix: /home/tscgs/gr-lora_sdr/build/usr/local
-- Building for version: ca195cb1 / 1.0.0git
-- No C++ unit tests... skipping
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so  
Traceback (most recent call last):
  File "<string>", line 2, in <module>
TypeError: Unicode-objects must be encoded before hashing
CMake Error at cmake/Modules/UseSWIG.cmake:199 (add_executable):
  The target name "/home/tscgs/gr-lora_sdr/build/swig/.cpp" is reserved or
  not valid for certain CMake features, such as generator expressions, and
  may result in undefined behavior.
Call Stack (most recent call first):
  cmake/Modules/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
  cmake/Modules/GrSwig.cmake:173 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:49 (GR_SWIG_MAKE)

CMake Error at cmake/Modules/GrSwig.cmake:211 (GR_LIBTOOL):
  Unknown CMake command "GR_LIBTOOL".
Call Stack (most recent call first):
  swig/CMakeLists.txt:54 (GR_SWIG_INSTALL)

-- Configuring incomplete, errors occurred!
See also "/home/tscgs/gr-lora_sdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/tscgs/gr-lora_sdr/build/CMakeFiles/CMakeError.log".

tapparelj, 2022-04-14:
The module has only been tested with GNU Radio 3.8.2, and will not work on GNU Radio 3.9 or 3.10 without porting it. The easiest way to make it work is to duplicate the conda environment described in the environment.yml file.

SainathVam5hi, 2022-04-14:
can you help/guide me with the installation of the environment I've tried installing it into python but cmake of gr-lora_sdr didn't work of course... can you please help me with specifying a location for the environment?

tapparelj, 2022-04-14:
You first need to have conda installed (e.g. by following the miniconda installation [here](https://docs.conda.io/projects/conda/en/latest/user-guide/install/linux.html#installing-on-linux) 
Then you can create an environment based on the file environment.yml in the root of this repository (instructions [here](https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html#id5))
Finally, when you want to use this environment you can activate it by typing `conda activate gr38` in a terminal. This environment contains all libraries to build the OOT module (e.g. GNU Radio, swig,cmake,...). Once activated in the terminal, you can build the OOT module by following the same process 'cmake.., make, make install'. if you want to activate the environment automatically when opening a new terminal, you can append the line `conda activate gr38` to the file ~\.bashrc
When the environment is activated in your terminal you can open gnuradio-companionby running `gnuradio-companion` in the terminal if you want the graphical view of the blocks.

SainathVam5hi, 2022-04-14:
I have installed conda and created an env with all the packages installed from gr38(renamed to lora in .yml) but still cannot install the OOT Module also installed rpp0's gr-lora but module not visible in the GUI

tscgs@tscgs:~/gr-lora_sdr-master$ conda env create -f environment.yml
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.10.3
  latest version: 4.12.0

Please update conda by running

    $ conda update -n base -c defaults conda

Downloading and Extracting Packages
gts-0.7.6            | 411 KB    | ##################################### | 100% 
libsodium-1.0.18     | 366 KB    | ##################################### | 100% 
libopenblas-0.3.12   | 8.9 MB    | ##################################### | 100% 
pyqt-impl-5.12.3     | 5.9 MB    | ##################################### | 100% 
pycparser-2.20       | 94 KB     | ##################################### | 100% 
alsa-lib-1.2.3       | 560 KB    | ##################################### | 100% 
libnghttp2-1.43.0    | 808 KB    | ##################################### | 100% 
python-3.6.13        | 38.4 MB   | ##################################### | 100% 
gnuradio-qtgui-3.8.2 | 1.3 MB    | ##################################### | 100% 
libstdcxx-ng-9.3.0   | 4.0 MB    | ##################################### | 100% 
libiconv-1.16        | 1.4 MB    | ##################################### | 100% 
swig-3.0.12          | 1.2 MB    | ##################################### | 100% 
nspr-4.30            | 233 KB    | ##################################### | 100% 
xorg-libxdmcp-1.1.3  | 19 KB     | ##################################### | 100% 
glew-2.1.0           | 647 KB    | ##################################### | 100% 
ca-certificates-2020 | 137 KB    | ##################################### | 100% 
libblas-3.9.0        | 11 KB     | ##################################### | 100% 
zeromq-4.3.4         | 352 KB    | ##################################### | 100% 
mpir-3.0.0           | 730 KB    | ##################################### | 100% 
pcre-8.44            | 261 KB    | ##################################### | 100% 
pyqtwebengine-5.12.1 | 175 KB    | ##################################### | 100% 
kernel-headers_linux | 707 KB    | ##################################### | 100% 
gcc_linux-64-9.3.0   | 23 KB     | ##################################### | 100% 
libtiff-4.2.0        | 633 KB    | ##################################### | 100% 
jpeg-9d              | 264 KB    | ##################################### | 100% 
liblapack-3.9.0      | 11 KB     | ##################################### | 100% 
graphite2-1.3.13     | 102 KB    | ##################################### | 100% 
icu-68.1             | 13.0 MB   | ##################################### | 100% 
six-1.15.0           | 14 KB     | ##################################### | 100% 
ncurses-6.2          | 985 KB    | ##################################### | 100% 
fonts-conda-forge-1  | 4 KB      | ##################################### | 100% 
liblimesuite-20.10.0 | 401 KB    | ##################################### | 100% 
libcurl-7.76.0       | 328 KB    | ##################################### | 100% 
curl-7.76.0          | 149 KB    | ##################################### | 100% 
hicolor-icon-theme-0 | 14 KB     | ##################################### | 100% 
libgd-2.3.2          | 299 KB    | ##################################### | 100% 
qwt-6.1.6            | 4.0 MB    | ##################################### | 100% 
markupsafe-1.1.1     | 27 KB     | ##################################### | 100% 
limesuite-20.10.0    | 2.4 MB    | ##################################### | 100% 
patchelf-0.11        | 78 KB     | ##################################### | 100% 
portaudio-19.6.0     | 130 KB    | ##################################### | 100% 
uhd-3.15.0.0         | 8.5 MB    | ##################################### | 100% 
soapysdr-module-uhd- | 225 KB    | ##################################### | 100% 
binutils_impl_linux- | 9.3 MB    | ##################################### | 100% 
pysocks-1.7.1        | 27 KB     | ##################################### | 100% 
libstdcxx-devel_linu | 13.9 MB   | ##################################### | 100% 
_libgcc_mutex-0.1    | 3 KB      | ##################################### | 100% 
libxml2-2.9.10       | 1.3 MB    | ##################################### | 100% 
gnuradio-video-sdl-3 | 193 KB    | ##################################### | 100% 
lxml-4.6.3           | 1.5 MB    | ##################################### | 100% 
ninja-1.10.2         | 2.4 MB    | ##################################### | 100% 
pyqt5-sip-4.19.18    | 310 KB    | ##################################### | 100% 
gxx_impl_linux-64-9. | 10.7 MB   | ##################################### | 100% 
fonts-conda-ecosyste | 4 KB      | ##################################### | 100% 
xorg-libxau-1.0.9    | 13 KB     | ##################################### | 100% 
[... 148 lines cut ...]
After activating env error sill exists:

(lora) tscgs@tscgs:~$ cd gr-lora_sdr-master
(lora) tscgs@tscgs:~/gr-lora_sdr-master$ cd build
(lora) tscgs@tscgs:~/gr-lora_sdr-master/build$ cmake ..
-- Conda installed GNU Radio. Setting CMAKE_INSTALL_PREFIX to /home/tscgs/anaconda3/envs/lora
-- Build type not specified: defaulting to release.
-- Using GMP.
-- User set python executable /usr/bin/python3
[... cut ...]
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so  
CMake Error at cmake/Modules/GrSwig.cmake:211 (GR_LIBTOOL):
  Unknown CMake command "GR_LIBTOOL".
Call Stack (most recent call first):
  swig/CMakeLists.txt:54 (GR_SWIG_INSTALL)

-- Configuring incomplete, errors occurred!
See also "/home/tscgs/gr-lora_sdr-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/tscgs/gr-lora_sdr-master/build/CMakeFiles/CMakeError.log".
(lora) tscgs@tscgs:~/gr-lora_sdr-master/build$

tapparelj, 2022-04-14:
Very strange indeed. The error  Unknown CMake command "GR_LIBTOOL". seems to be linked to the use of gnuradio 3.9-3.10 with modules made for 3.8. It looks like cmake uses some libraries outside the conda environment.
What are the output of the commands `gnuradio-config-info -v --prefix  --sysconfdir --prefsdir --userprefsdir --enabled-components --cc --cxx --cflags` and `libtool --version` in the lora environment?
Can you also provide the content of "/home/tscgs/gr-lora_sdr-master/build/CMakeFiles/CMakeOutput.log"?

SainathVam5hi, 2022-04-14:
output of the commands requested:

(lora) tscgs@tscgs:~$ gnuradio-config-info -v
3.8.2.0
(lora) tscgs@tscgs:~$ gnuradio-config-info --prefix
/home/tscgs/anaconda3/envs/lora/.
(lora) tscgs@tscgs:~$ gnuradio-config-info --sysconfdir
/home/tscgs/anaconda3/envs/lora/etc
(lora) tscgs@tscgs:~$ gnuradio-config-info --prefsdir
/home/tscgs/anaconda3/envs/lora/etc/gnuradio/conf.d
(lora) tscgs@tscgs:~$ gnuradio-config-info --userprefsdir
/home/tscgs/.gnuradio
(lora) tscgs@tscgs:~$ gnuradio-config-info --enabled-components
python-support;gnuradio-runtime;gr-ctrlport;gnuradio-companion;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;* portaudio;gr-channels;gr-qtgui;gr-trellis;gr-uhd;gr-utils;gr_modtool;gr-video-sdl;gr-vocoder;* codec2;* freedv;gr-wavelet;gr-zeromq
(lora) tscgs@tscgs:~$ gnuradio-config-info --cc
x86_64-conda-linux-gnu-cc (crosstool-NG 1.24.0.133_b0863d8_dirty) 9.3.0 
Copyright (C) 2019 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
(lora) tscgs@tscgs:~$ gnuradio-config-info --cxx
x86_64-conda-linux-gnu-c++ (crosstool-NG 1.24.0.133_b0863d8_dirty) 9.3.0 
Copyright (C) 2019 Free Software Foundation, Inc. 
This is free software see the source for copying conditions.  There is NO 
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
(lora) tscgs@tscgs:~$ gnuradio-config-info --cflags
/home/conda/feedstock_root/build_artifacts/gnuradio_1614952867556/_build_env/bin/x86_64-conda-linux-gnu-cc:::-O3 -DNDEBUG -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/tscgs/anaconda3/envs/lora/include -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/gnuradio_1614952867556/work=/usr/local/src/conda/gnuradio-3.8.2.0 -fdebug-prefix-map=/home/tscgs/anaconda3/envs/lora=/usr/local/src/conda-prefix -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized 
/home/conda/feedstock_root/build_artifacts/gnuradio_1614952867556/_build_env/bin/x86_64-conda-linux-gnu-c++:::-O3 -DNDEBUG -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/tscgs/anaconda3/envs/lora/include -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/gnuradio_1614952867556/work=/usr/local/src/conda/gnuradio-3.8.2.0 -fdebug-prefix-map=/home/tscgs/anaconda3/envs/lora=/usr/local/src/conda-prefix -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
(lora) tscgs@tscgs:~$ libtool --version
libtool (GNU libtool) 2.4.6
Written by Gordon Matzigkeit, 1996

Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
(lora) tscgs@tscgs:~$ 

CMakeOutput.log:
[CMakeOutput.log](https://github.com/tapparelj/gr-lora_sdr/files/8489819/CMakeOutput.log)

tapparelj, 2022-04-14:
did you cleared the content of the build directory before running `cmake ..`? some path leads outside the environment

SainathVam5hi, 2022-04-14:
No,I don't think so. how do i do that?

tapparelj, 2022-04-14:
just delete the content of the build folder before running cmake again

tapparelj, 2022-04-14:
Glad to hear that! Let me know if you have other issues

SainathVam5hi, 2022-04-14:
This may not be a issue related to this rep but I'm finding it difficult to add soapy to the gnu env. can you help me out?

SainathVam5hi, 2022-04-17:
Hey I'm using the blocks that your colleague had developed since it supports GNU3.9.
Though i was able to build install and run the block I'm not able to receive on SX126x
----------------------------------------------------------
[INFO] Opening HackRF One #0 f77c60dc226275c3...
set_min_output_buffer on block 5 to 10000000
set_min_output_buffer on block 6 to 10000000
set_min_output_buffer on block 8 to 10000000
set_min_output_buffer on block 9 to 10000000
set_min_output_buffer on block 11 to 10000000
set_min_output_buffer on block 12 to 10000000
set_min_output_buffer on block 10 to 10000000
set_min_output_buffer on block 7 to 10000000
gr::log :INFO: data_source0 - INFO:Processing frame [:0/10]
----------------------------------------------------------------------------------
Below is my current block config.

[image]

tapparelj, 2022-04-25:
Hi,
It's quite difficult to debug it as I don't know what colleague you make reference to. From what I see here, the sync words parameter is strange. I tested with the SX1276 chips and they use the sync words 0x12 by default. The SX126x defines the sync words differently like 0x1424 which I suppose should match the value 0x12 set in your Tx block.
Aside from that, I don't know how they configures the bandwidth and os_factor in the "Modulate" block as the bw should be the one defined for the LoRa transmission, and the os_factor should match the ration between hackRf sample rate and the bw.
It might be easier to contect the person that made the hierarchical block "Tx" to get more info.

closed 2022-06-10 by tapparelj