tapparelj/gr-lora_sdr #15: unexpected EOF while parsing

MartinaS1u9a, 2022-05-02:
Heyhey, 
I'm very new to this. But I managed to install gr-lora_sdr on a Linux Virtual Machine using Ubuntu 20.04. 
But some of those Blocks in GnuRadio don't show the options one can choose from. 
"add header" is one of them. As soon as I open it up to specifiy the options the poped up window stays blank. 
I tried installing it again and even unpacked it again. But somehow it still doesn't work. 
I'm don't want to delete the one version that I got running on GnuRadio, since it took me some days and a lot of breakdowns for the blocks to be shown in GnuRadio, and I am a bit at my witts ends. 
I hope someone can help me please. I will be forever grateful for it.

tapparelj, 2022-05-03:
Hello,
I'll try my best to help you make it work. Can you provide more info such as:
- Which version of gnuradio are you using? 
- Can you specify one or two more blocks that appears incorrectly and some that show up correctly? That would help narrowing down possible causes of this issue.
- The Issue title mention a parsing error, can you provide the full error description?
-  What happens if you directly runs a flowgraph script in the terminal? (go in the location gr-lora_sdr/apps/single_user and execute `python3 tx_rx_simulation.py `)

SlumiJr, 2022-05-07:
Hello,

sorry for the late response, had to do some other work. 
I was using gnuradio 3.8.1. 
Could that be the issue? My Professor thought that there wouldn't be so many differences between 3.8.1 and 3.8.2., since I'm working on a system my university provided for me, so I was trying to get it to work an that. 
I couldn't specifiy blocks like fft_demod, interleaver and frame_sync.
Maybe if I try to update the gnuradio version to 3.8.2 it could solve the issue?

tapparelj, 2022-05-07:
It would be easier to answer if you could provide an answer to the few points I mentioned previously.
I also don't think the version 3.8.1 instead of 3.8.2 is the main issue. If you can use  conda, the easiest way to make it work is to create a environment based on the yml file provided in the root of this repo

SlumiJr, 2022-06-06:
Heyhey, 

I just wanted to say, I managed to resolve that issue. After some trial and error, I decided to install gnuradio on a completely new virtual machine, since i've been using the one we had on our university computers. After installing it with pybombs i didn't get that issue again. 
But another question popped up, while trying out some stuff with LoRa blocks: The examples that are provided (tx_rx_simulation) are using an interpolating fir filter and I wanted to ask, what those taps are meaning? 
I get that it's some kind of low-pass filter, but I do not completely understand that filter and what kind of transfer function is used?

tapparelj, 2022-06-07:
Hello,
This is indeed a low-pass filter used in conjuction with upsampling to provide a signal oversampled by a factor 4 to the synchronization block. This oversampling can also be done with a higher sampling rate of the receiver if you prefer. It main purpose is to allow to compensate the sampling time offset.
The taps were generated with the `intfilt(ovesampling=4,non-zero_values=2,bandwidth_occupation=1)` function of MATLAB.

closed 2022-07-02 by tapparelj