tapparelj/gr-lora_sdr #17: swig error

phileas29, 2022-05-16:
I have this message from gnu radio and from python command line when `import lora_sdr` :
```
from .lora_sdr_swig import *
ModuleNotFoundError: No module named 'lora_sdr.lora_sdr_swig'
```

phileas29, 2022-05-16:
installation process as describe in readme not to seem produce swig files

tapparelj, 2022-05-18:
Hi,
Can you provide some jnformation about the installation of gnuradio? Such as the version, how you installed it (with pybombs, conda, from source,...), which os are you using.

squid256, 2022-05-19:
I am also getting this error in gnuradio-companion. I was able to find the lora_sdr_swig.i file, but it wasn't in the gnuradio/swig folder with other similar files. I tried copying it over, but it didn't seem to change anything. I am running GNU Radio 3.8.1 on a remote computer that is using Ubuntu 18.04. I'm not sure how it was installed

tapparelj, 2022-05-20:
There might be an issue with you PYTHONPATH and LD_LIBRARY_PATH. When you build the OOT module and run the ´make install´, where are the actual files copied in particular the file lib/libgnuradio-lora_sdr.so and lora_sdr/_lora_sdr_swig.so? 
You might need to specify those two path as indicated [here](https://wiki.gnuradio.org/index.php/ModuleNotFoundError) using the export command.
Note that you have to use the export commands in the same terminal prior to launching gnuradio-companion with the command `gnuradio-companion`, or directly executing a flowgraph with `python3 {nameOfTheFlowgraph}.py`.

squid256, 2022-05-20:
I had tried doing that and it didn't seem to fix it. I have just gotten it to work, though. I completely restarted and just created the conda environment before running anything and it looks like it worked. I think there must have been something that was installed improperly so when I tried switching to the conda environment it wasn't able to overwrite it. Here are the exact steps I used, and it seems to have worked.
1. Install Anaconda
    cd /tmp
    touch output.txt
    curl https://repo.anaconda.com/archive/Anaconda3-2022.05-Linux-x86_64.sh > output.txt
    bash output.txt
      
2. Follow the instructions from the anaconda installation (including exiting and reopening the terminal)
 
3. Clone the repository.
    cd [directory where you want the repository]
    git clone https://github.com/tapparelj/gr-lora_sdr.git

5. Initialize the conda environment provided.
    cd gr-lora_sdr
    conda env create -f environment.yml
    
6. Activate the conda environment.
    conda activate gr38
    
7. Build the repository.
    mkdir build
    cd build
    export $GR_INSTALL_PREFIX="gnuradio-config-info --prefix"
    cmake .. -DCMAKE_INSTALL_PREFIX=$GR_INSTALL_PREFIX
    make
    sudo make install
    sudo ldconfig

PeteK6TJ, 2022-06-01:
In step 6 3rd line, is there really a $ and really a --prefix?  It worked when I omitted these two items:
export GR_INSTALL_PREFIX="gnuradio-config-info"

squid256, 2022-06-09:
Sorry, yeah, that $ shouldn't be there in line 3. --prefix should be there though if you're putting the install prefix in the -DCMAKE_INSTALL_PREFIX part. The "gnuradio-config-info" command prints out a help menu if you don't have any options, at least for the versions of gnuradio that I'm using. The cmake .. command can run just fine without the -DCMAKE... part, but sometimes it isn't configured correctly when you don't put the install prefix.

squid256, 2022-06-09:
I can't replicate what I did right now, so I'm not sure what I did differently. What you changed seems to be working for me

tapparelj, 2022-08-03:
I close this issue as we moved to gnuradio 3.10 which doesn't use swig anymore.

closed 2022-08-03 by tapparelj