tapparelj/gr-lora_sdr #18: Feature Request: Frame sync outputs timing

PeteK6TJ, 2022-06-01:
IIUC, One can use eg the Semtech chip to decode, but not to get ranging.  Could the Frame sync block output the time of the beginning of the network identifier??

tapparelj, 2022-06-10:
If you want you can add a few lines of code at line 474 of lib/frame_sync_impl.cc to get the current time and write it in a file for example. This should give you a rough estimation of the time of the beginning of the payload. However, this method will probably give you a low precision since it would be dependent on the processing schedule of your CPU.
We will try to add that information in the future, but only if we can get a precision that is meaningful.

closed 2022-06-10 by tapparelj

lemarquez, 2022-06-28:
Best regards
I have tried to modify lines of code in frame_sync_impl.cc but the change does not take effect when executing the Lora_RX code in the GNU radio. What suggestion could you give me in this matter?

tapparelj, 2022-06-29:
Hi,
If you modify the source code, you have to run the `make install` command again while in the build folder (similar to the first time you build the module).

tapparelj, 2022-06-29:
Glad to hear it! 
Don't hesitate if you have other issues