tapparelj/gr-lora_sdr #26: No example code is executed with USRP

lemarquez, 2022-08-23:
**After following the installation steps for the latest version of the gr-lora-sdr code (Gnuradio 3.10), I have not been able to run any of the codes in the "Examples" folder, except tx_rx_functionnality_check.grc.
From the command line I execute uhd_find_devices and the result is:**

[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.HEAD-release
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 3166C2F
    name: MyB200
    product: B200
    type: b200

**When I run uhd_usrp_probe the output is:**
INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.HEAD-release
[INFO] [B200] Detected Device: B200
Error: EnvironmentError: IOError: empty input file /home/lora/miniconda3/envs/gr310

**It only works if I run it as sudo uhd_usrp_probe, where the output is:**
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.3.0.0-144-gdfc7b31c
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
  _____________________________________________________
 /
|       Device: B-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: B200
|   |   serial: 3166C2F
|   |   name: MyB200
|   |   product: 1
|   |   revision: 5
|   |   FW Version: 8.0
|   |   FPGA Version: 16.0
|   |   
|   |   Time sources:  none, internal, external, gpsdo
|   |   Clock sources: internal, external, gpsdo
|   |   Sensors: ref_locked
|   |     _____________________________________________________
|   |    /
|   |   |       RX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       RX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Frontend: A
|   |   |   |   Name: FE-RX1
|   |   |   |   Antennas: TX/RX, RX2
|   |   |   |   Sensors: temp, rssi, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       RX Codec: A
|   |   |   |   Name: B200 RX dual ADC
|   |   |   |   Gain Elements: None
|   |     _____________________________________________________
|   |    /
|   |   |       TX DSP: 0
|   |   |   
|   |   |   Freq range: -8.000 to 8.000 MHz
|   |     _____________________________________________________
|   |    /
|   |   |       TX Dboard: A
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Frontend: A
|   |   |   |   Name: FE-TX1
|   |   |   |   Antennas: TX/RX
|   |   |   |   Sensors: temp, lo_locked
|   |   |   |   Freq range: 50.000 to 6000.000 MHz
|   |   |   |   Gain range PGA: 0.0 to 89.8 step 0.2 dB
|   |   |   |   Bandwidth range: 200000.0 to 56000000.0 step 0.0 Hz
|   |   |   |   Connection Type: IQ
|   |   |   |   Uses LO offset: No
|   |   |     _____________________________________________________
|   |   |    /
|   |   |   |       TX Codec: A
|   |   |   |   Name: B200 TX dual DAC
|   |   |   |   Gain Elements: None

**From gnuradio-companion, when I run tx_rx_simulation the output is:**
Generating: '/home/lora/.grc_gnuradio/tx_rx_simulation.py'
Executing: /home/lora/miniconda3/envs/gr310/bin/python -u /home/lora/.grc_gnuradio/tx_rx_simulation.py

**And if I run lora_RX the error comes out:**
Generating: '/home/lora/gr-lora_sdr/examples/lora_RX.py'
Executing: /home/lora/miniconda3/envs/gr310/bin/python -u /home/lora/gr-lora_sdr/examples/lora_RX.py
[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.HEAD-release
[INFO] [B200] Detected Device: B200
Traceback (most recent call last):
  File "/home/lora/gr-lora_sdr/examples/lora_RX.py", line 250, in <module>
    main()
  File "/home/lora/gr-lora_sdr/examples/lora_RX.py", line 228, in main
    tb = top_block_cls()
  File "/home/lora/gr-lora_sdr/examples/lora_RX.py", line 93, in __init__
    self.uhd_usrp_source_0 = uhd.usrp_source(
RuntimeError: EnvironmentError: IOError: empty input file /home/lora/miniconda3/envs/gr310

**I have done all the previous USB and UHD configuration steps, as recommended at: https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux
Still the error persists.
Please if you can guide me to determine the cause of the error.**

tapparelj, 2022-08-24:
Hello @lemarquez,
- First regarding `tx_rx_simulation`:
    - Did you update the path of the `File Source` block?
    - If you execute directly in a terminal using `python3 tx_rx_simulation.py` anything else get printed in the terminal? (be sure that the option of the block `CRC verif` is set to Yes)
- Second for the USRP issue:
    - It seems that the configuration of the USB is not set up properly. But with sudo right it's fine. so can you try the following:
    - Can you execute the python script as sudo e.g. `sudo python3 lora_RX.py` ?
    - What if you use a sudo terminal with `sudo -s` and then ` python3 lora_RX.py`?
    - Based on [This](https://wiki.gnuradio.org/index.php/CondaInstall#Building_GNU_Radio_from_source_within_a_conda_environment) You can try to add libudev to the conda environment `conda install -c conda-forge libudev-devel-cos6-x86_64`  and then remove ~conda/envs/gr310/lib/libudev.so.0

lemarquez, 2022-08-24:
**Thank you for your answer.

- About tx_rx_simulation.py, I had not made the path change to the File Source. 

After setting the proper path and executing, the output is as follows:**

mmse_resampler_cc :info: set_min_output_buffer on block 17 to 563
fir_filter_blk<IN_T,OUT_T,TAP_T> :info: set_min_output_buffer on block 18 to 563
multiply_cc :info: set_min_output_buffer on block 22 to 563
sig_source :info: set_min_output_buffer on block 21 to 563
add_cc :info: set_min_output_buffer on block 19 to 563
fastnoise_source :info: set_min_output_buffer on block 20 to 563
Press Enter to quit: rx msg: HztlXcpsegzpnbkq�3�4�-�t
                                                     Dg�!1o��s9�ʋ�<p�B��i��_��r���t�ID�*�T��~W8��s�}���}�2��ێ�_��9b�D�q��"�K�^N����k/ �ԇi=0%o�R欁�0#b����U(�u�ۦmݳ|+W��������Iv(��+�Ĩă[L�
                        ɩ��
K����=!�]

CRC invalid

**If I change the message in the example_tx_source.txt file, the receiver is put on hold, as shown in the following image**
[image]

**- About USRP, I tried sudo python3 lora_RX.py and the following was the output.**
[image]

Likewise, when I run sudo -s

[image]

**Then, I successfully ran conda install -c conda-forge libudev-devel-cos6-x86_64
When trying to remove ~conda/envs/gr310/lib/libudev.so.0 the result is**
E: Unable to locate package libudev.so.0
E: Couldn't find any package by glob 'libudev.so.0'
E: Couldn't find any package by regex 'libudev.so.0'

**The file 'libudev.so.0' does not exist in the lib folder, the one that exists is libudev.so.1, should this one be deleted?**

**After the suggested installation, I run again lora_rx.grc and it stays indefinitely trying to access to the usrp image, as shown in the screenshot below**
[image]

lemarquez, 2022-08-24:
**[image]**

tapparelj, 2022-08-25:
Hello,
- For the simulation:
    - it looks to me that you chose to use the implicit header mode. If you use it, you should provide the size of the payload to the receiver. By default I use the explicit header mode and leave the payload lentgh to a default 255, since it doesn't matter in explicit header mode.
    - When you change the message, do you add a terminating "," ?
- For the USRP:
    - When using sudo it fails to find the module as it overwrite the different variables set when activating the conda environment. We have to make the USRP work without sudo then.
    - So it's not that library that was missing.
    - I think we should focus on making uhd_usrp_probe work as it indicates an issue with the UHD configuration itself.
        - What is the output of `uhd_usrp_probe --args="type=b200"`, of `uhd_find_devices`, and `uhd_find_devices --args="type=b200"`  ?
        - the [INFO] stops before [INFO] [B200] Operating over USB 3 so the problem probably comes from the USB configuration specific to Conda, can you look a bit into that (I'm not familiar to USRP over USB, but I never had issues with LimeSDR also over USB).

lemarquez, 2022-08-25:
**Hello, Thank you again!**

**- tx_rx_simulation:**
With explicit header mode and by adding the terminating "," we can successfully receive the message.
[image]

**- USRP:**

- The output of uhd_usrp_probe --args="type=b200" is

[image]

- The outputs of "uhd_find_devices" and "uhd_find_devices --args="type=b200"" are the same:
[image]

- Finally, the error was solved using uhd_images_downloader without sudo in the gr310 environment. Now the lora_rx is working
[image]

**Thank you very much for all your help**

tapparelj, 2022-08-26:
Glad to hear that!
Let me know if you encounter any other issue.

closed 2022-08-26 by tapparelj