tapparelj/gr-lora_sdr #27: gnuradio version conda env

lahc1, 2022-09-01:
after installing conda env:  `conda env create -f environment.yml `, and `cmake .. -DCMAKE_INSTALL_PREFIX=/home/ubuntu16-sdr2/anaconda3/envs/gr310/` in gr310 env
gnuradio 3.8.2.0 is still running:
```
(gr310) ubuntu16-sdr2@ubuntu16sdr2-OptiPlex-7070:~/gr-lora_sdr/build$ gnuradio-companion -h
usage: gnuradio-companion [-h] [--log {debug,info,warning,error,critical}] [flow_graphs ...]

GNU Radio Companion 3.8.2.0 This program is part of GNU Radio GRC comes with ABSOLUTELY NO WARRANTY. This is free software, and you are welcome to redistribute it.

positional arguments:
  flow_graphs

options:
  -h, --help            show this help message and exit
  --log {debug,info,warning,error,critical}
```
```
(gr310) ubuntu16-sdr2@ubuntu16sdr2-OptiPlex-7070:~/gr-lora_sdr/examples$ gnuradio-companion lora_TX.grc 
<<< Welcome to GNU Radio Companion 3.8.2.0 >>>

Block paths:
	/home/ubuntu16-sdr2/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/ubuntu16-sdr2/gr-lora_sdr/examples/lora_TX.grc"
Segmentation fault (core dumped)
```

any suggestions how to solve this issue ?

tapparelj, 2022-09-01:
Hello @lahc1,
It seems the some paths to the older version of gnuradio are still remaining after the activation of the new conda environment. Can you answer to the following questions:
- How did you installed gnuradio 3.8.2? with conda, pybombs, apt, from source, or any other way?
- While in the environment gr310
    - Do the result `conda list | grep "gnuradio"` only mention versions 3.10.3.0?
    - What is the result of `gnuradio-config-info -v --prefix`?
    - What is the result of `echo $PATH `

lahc1, 2022-09-01:
Hi @tapparelj,
yes it seems a conda path problem, but I am not able to solve it.
I installed gnuradio 3.8.2 using conda. 
_(gr310) ubuntu16-sdr2@ubuntu16sdr2-OptiPlex-7070:~/gr-lora_sdr_folder/gr-lora_sdr/examples$ **conda list | grep "gnuradio"**_
```
gnuradio                  3.10.3.0        py310hb890b00_2    conda-forge
gnuradio-core             3.10.3.0        py310h179aa5a_2    conda-forge
gnuradio-grc              3.10.3.0        py310h0dc8b28_2    conda-forge
gnuradio-iio              3.10.3.0        py310h3058d4e_2    conda-forge
gnuradio-qtgui            3.10.3.0        py310hc030b51_2    conda-forge
gnuradio-soapy            3.10.3.0        py310h00a44fc_2    conda-forge
gnuradio-uhd              3.10.3.0        py310he2bccb3_2    conda-forge
gnuradio-video-sdl        3.10.3.0        py310h0dc8b28_2    conda-forge
gnuradio-zeromq           3.10.3.0        py310h935459f_2    conda-forge
```
_(gr310) ubuntu16-sdr2@ubuntu16sdr2-OptiPlex-7070:~/gr-lora_sdr_folder/gr-lora_sdr/examples$ **gnuradio-config-info -v --prefix**_
```
/home/ubuntu16-sdr2/anaconda3/envs/gr310/
3.10.3.0
```
_(gr310) ubuntu16-sdr2@ubuntu16sdr2-OptiPlex-7070:~/gr-lora_sdr_folder/gr-lora_sdr/examples$ **echo $PATH**_
```
/home/ubuntu16-sdr2/bin:/home/ubuntu16-sdr2/.local/bin:/home/ubuntu16-sdr2/anaconda3/envs/gr310/bin:/home/ubuntu16-sdr2/anaconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
```
_(gr310) ubuntu16-sdr2@ubuntu16sdr2-OptiPlex-7070:~/gr-lora_sdr_folder/gr-lora_sdr/examples$ **python3 tx_rx_functionnality_check.py**_ 
```
Traceback (most recent call last):
  File "/home/ubuntu16-sdr2/gr-lora_sdr_folder/gr-lora_sdr/examples/tx_rx_functionnality_check.py", line 23, in <module>
    import gnuradio.lora_sdr as lora_sdr
ModuleNotFoundError: No module named 'gnuradio.lora_sdr'
```
any suggestions?

tapparelj, 2022-09-01:
Hi,
It seems that there is an issue with the overall installation if you can't launch gnuradio-companion. Can you try duplicating the conda environment again and when building the lora_sdr module, do not specify the prefix (it will install in the current conda environment if any is active)?

lahc1, 2022-09-06:
> Hi, It seems that there is an issue with the overall installation if you can't launch gnuradio-companion. Can you try duplicating the conda environment again and when building the lora_sdr module, do not specify the prefix (it will install in the current conda environment if any is active)?

Hi, I have done this but still stuck with same issues. I even deleted all the environment then install again and still stuck. any recommendation? thanks

tapparelj, 2022-09-06:
Hi,
The argument 0 of Gtk.StyleContext.add_provider_for_screen( is uspposed to be your screen. 
- Do you have a peculiar setup for your display? 
- What is the output of `echo $DISPLAY`?
   - if there is not output can you try to run `export DISPLAY=:0` ?

lahc1, 2022-09-06:
hi,
that is not an issue, terminal is trying to launch Xming. I tried it also on the PC itself and got same output of Gnuradio.

tapparelj, 2022-09-07:
Hi,
When you speak about the PC itself, just to be sure, its OS is Ubuntu?

lahc1, 2022-09-07:
> 

yes, it is Ubuntu 18

tapparelj, 2022-09-07:
Can you try to set the path `/home/ubuntu16-sdr2/anaconda3/envs/gr310/bin` in the very beginning of $PATH to be sure it uses those binaries first?
`export PATH=/home/ubuntu16-sdr2/anaconda3/envs/gr310/bin:$PATH`

lahc1, 2022-09-07:
[image]

after setting the path to:  `export PATH=/home/ubuntu16-sdr2/anaconda3/envs/gr310/bin:$PATH`
GNURadio is running but with version 3.8.2. 

```
(gr310) ubuntu16-sdr@ubuntu16sdr-OptiPlex-7070:~/Downloads/gr-lora_sdr/examples$ conda list | grep "gnuradio"
gnuradio                  3.10.3.0        py310hb890b00_2    conda-forge
gnuradio-core             3.10.3.0        py310h179aa5a_2    conda-forge
gnuradio-grc              3.10.3.0        py310h0dc8b28_2    conda-forge
gnuradio-iio              3.10.3.0        py310h3058d4e_2    conda-forge
gnuradio-qtgui            3.10.3.0        py310hc030b51_2    conda-forge
gnuradio-soapy            3.10.3.0        py310h00a44fc_2    conda-forge
gnuradio-uhd              3.10.3.0        py310he2bccb3_2    conda-forge
gnuradio-video-sdl        3.10.3.0        py310h0dc8b28_2    conda-forge
gnuradio-zeromq           3.10.3.0        py310h935459f_2    conda-forge
```
```
(gr310) ubuntu16-sdr@ubuntu16sdr-OptiPlex-7070:~/Downloads/gr-lora_sdr/examples$ gnuradio-config-info -v --prefix
/home/ubuntu16-sdr/miniconda3/envs/gr310/
3.10.3.0
```
```
(gr310) ubuntu16-sdr@ubuntu16sdr-OptiPlex-7070:~/Downloads/gr-lora_sdr/examples$ echo $PATH
/home/ubuntu16-sdr2/anaconda3/envs/gr310/bin:/home/ubuntu16-sdr/.local/bin:/home/ubuntu16-sdr/miniconda3/envs/gr310/bin:/home/ubuntu16-sdr/miniconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
```

tapparelj, 2022-09-07:
Do you have anything in `echo $PYTHONPATH` and `echo $LD_LIBRARY_PATH` ?

lahc1, 2022-09-07:
I already set echo `$PYTHONPATH` and `echo $LD_LIBRARY_PATH` as you mentioned. It is still not working.
```
(gr310) ubuntu16-sdr@ubuntu16sdr-OptiPlex-7070:~/Downloads/gr-lora_sdr/examples$ conda install gnuradio=3.10 python=3.10.5
Collecting package metadata (current_repodata.json): done
Solving environment: done

# All requested packages already installed.

Retrieving notices: ...working... done
```

tapparelj, 2022-09-07:
In a terminal what is the output from `python`, `import gnuradio`, `gnuradio` ?

lahc1, 2022-09-07:
It worked after following this page: https://wiki.gnuradio.org/index.php?title=ModuleNotFoundError#B._Finding_the_Python_library
```
(gr310) ubuntu16-sdr@ubuntu16sdr-OptiPlex-7070:~/Downloads/gr-lora_sdr/examples$ gnuradio-companion -h
usage: gnuradio-companion [-h] [--log {debug,info,warning,error,critical}] [flow_graphs ...]

GNU Radio Companion 3.10.3.0 This program is part of GNU Radio GRC comes with ABSOLUTELY NO WARRANTY. This is free software, and you are welcome to redistribute it.

positional arguments:
  flow_graphs

options:
  -h, --help            show this help message and exit
  --log {debug,info,warning,error,critical}
```

I have another problem with UHD, it got updated to version 4.2.0. But the version I need is 3.15 because the USRPs are also flashed with UHD3.15:
```
(gr310) ubuntu16-sdr@ubuntu16sdr-OptiPlex-7070:~/Downloads/gr-lora_sdr/examples$ ./lora_TX.py 
[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.HEAD-release
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.10.2,type=n3xx,product=n320,serial=31EABBA,claimed=False,addr=192.168.10.2
```

any suggestion how to switch back to UHD3.15?
thanks.

tapparelj, 2022-09-07:
Can't you just `conda remove uhd` and `conda install -c conda-forge uhd=3.15` ?

lahc1, 2022-09-07:
it worked, now I have UHD 3.15.0. But `uhd_usrp_sing `is missing from GNURadio
is there any way to make UHD 3.15 compatible with GNURadio 3.10.3 ?

[image]

tapparelj, 2022-09-08:
Hi,
is gnuradio-uhd still listed by `conda list | grep "gnuradio"` ? If not you can try to install it with `conda install gnuradio-uhd` and see if it tries to override your UHD installation. Else you might try to build gr-uhd from source (https://github.com/gnuradio/gnuradio/tree/main/gr-uhd). 
In any case, I'm not aware of which version of UHD is compatible or not with gnuradio 3.10 but reflashing your USRP might be the easiest way to go.

closed 2022-09-15 by tapparelj