tapparelj/gr-lora_sdr #31: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xcc in position 8: invalid continuation byte

EdikGres, 2022-10-21:
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/opt/homebrew/Cellar/gnuradio/3.10.4.0/lib/python3.10/site-packages/gnuradio/grc/gui/Executor.py", line 115, in run
    r = self.process.stdout.read(1)
  File "/opt/homebrew/Cellar/python@3.10/3.10.8/Frameworks/Python.framework/Versions/3.10/lib/python3.10/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xcc in position 8: invalid continuation byte

This exception is thrown if I receive a packet containing certain characters, but I am allowed to receive from 0x00 to 0xFF

MacOS

tapparelj, 2022-10-25:
Hello @EdikGres,
I never tried using gnuradio on MacOS, but it seems that _buffer_decode is using utf-8 instead of latin-1. Can you execute the scripts if you execute them directly in a terminal. (e.g `python tx_rx_functionnality_check.py` in the folder examples)

EdikGres, 2022-11-18:
everything was solved, I just rewrote the code a bit =)

closed 2022-11-18 by EdikGres