tapparelj/gr-lora_sdr #33: SX1278 cann't receive packets from USRP correctly with SF12

JaysonJie1111, 2022-11-04:
Hi, This is JaysonJie. I discuss the issue when I used the SX1278 to send a packet to usrp with SF 12, but I only received  a set of wrong characters。As shown in the figure below:
[image]
But, when I try to use other SFs (like 7,8,9,10,11)  to send messages through SX1278, usrp can receive the correct message。As shown in the figure below:
[image]
So, how can i deal with this problem?

(By the way, i also try to use usrp to send the message to SX1278 with SF 12, It didn't work. But in other SF cases, it worked! )

tapparelj, 2022-11-04:
Hi @JaysonJie1111,
I suspect that it is related to a mismatch of the setting of low datarate optimisation (LDRO). 
What bandwidth are you using?
By the way is it normal that you send two special characters in the payload beginning?

JaysonJie1111, 2022-11-04:
I have used 433.0MHz and 470.0MHz  in bandwidth.
These two special characters is the rf95 header in the payload beginning of SX1278.

tapparelj, 2022-11-04:
- Are you speaking of the carrier frequency ? because LoRa  usually uses a bandwidth between 125kHz and 500kHz.
- I see for the symbols. There is a block prepending the radiohead header to the payload that you can use when transmitting. You can insert it just before the whitening block (the destination can be set to 255 for broadcast).

JaysonJie1111, 2022-11-04:
- Sorry, I confused the bandwidth  with the  carrier frequency. I used 250kHz in bandwidth. And the central carrier frequency is 433.0MHz and 470.0MHz.
- Yes, when I use USRP to send packets to SX1278, I use this block just like you said. It made the usrp send the packet successfully in SF 7,8,9,10,11. but it also didn't work in SF12.
-

tapparelj, 2022-11-04:
LDRO should be enabled when a symbol duration exceed 16ms. Since you use a bandwidth of 250kHz and sf12, you have a symbol duration of $\frac{2^{12}}{250}=16.384ms$. 
Some libraries don't respect the 16ms criterion so you can either:
-  Enable it yourself by setting the register RegModemConfig3 (0x26) (see [datasheets](https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf) page 114 ).
- Disable it by changing the parameter LDRO in the interleaver block to "Disable"

JaysonJie1111, 2022-11-06:
Thanks! I found that we were using an older version of the tapparelj/gr-lora_sdr which is without low datarate optimisation support.
[image]
And then we tried to disable the LDRO function on our COTS node.
[image]
Finally, it work!!!  The USRP can receive the information from the COTS node correctly under the SF = 12 condition.
The COTS node also can receive the information from the USRP correctly under the SF = 12 condition.

We will update the **tapparelj/gr-lora_sdr** we used to the latest version to avoid this problem again.

tapparelj, 2022-11-07:
Glad to hear it!
I'm closing this issue then.

closed 2022-11-07 by tapparelj