tapparelj/gr-lora_sdr #34: Set preamble length

Radiomix2000, 2022-11-24:
The LoRa has such parameter as preamble length. How to set preamble length in symbols? Does the code support preamble setting?

tapparelj, 2022-11-25:
Hi @Radiomix2000 ,
I just pushed a new version that allow to choose the preamble length in the `modulate` and `frame_sync` blocks. 
You might have to use :
`make uninstall`  
`make clean`
before building the new version to avoid conflicts of constructor definition for those two blocks. 
Let me know if you encounter any issue.

Radiomix2000, 2022-11-25:
Thank you a lot!
 By the way a question about CRC. These standard LoRa modules like SX126x, SX127x have CRC settings (CRC enabled/disabled and CRC 1 or 2, what means CRC size 1 or 2 bytes) I'm not sure, does that means header CRC and payload CRC, like CRC 1 byte means only header CRC enabled, and CRC 2 means both header and payload CRC enabled? Or this CRC 1 or 2 bytes setting is absolutely different setting? If so, is it possible to implement that 1 or 2 bytes CRC in gr-lora-sdr as well?

tapparelj, 2022-11-25:
I am aware of functioning of the enable/disable payload CRC, but not of the CRC 1 or 2 that you are refering to. 
Could you point me toward where it is described in the [datasheets](https://www.mouser.com/datasheet/2/761/sx1276-1278113.pdf)?

Radiomix2000, 2022-11-25:
I found mention about 2 bytes CRC, but it's about FSK mode, not LoRa. So, it's not applicable to LoRa:
[image]

But here I found mention about 16-bit CRC for Lora: 
[image]
And again there is no apparent description, how to set it up..

tapparelj, 2022-11-25:
From what I have observed, LoRa can only enable or disable the payload CRC. The explicit header has a 5 bits checksum that cannot be disabled.
The payload CRC is based on CRC-16-CCITT with a slight modification, namely the last 2 payload bytes are XORed with the CRC obtained using the standard CCITT-16 on the rest of the payload.
If you want more information about how that information was found you can have a look at [this](https://www.epfl.ch/labs/tcl/wp-content/uploads/2020/02/Reverse_Eng_Report.pdf) reverse engineering report, section 2.5.

Radiomix2000, 2022-11-25:
Yes, thank you. I had already read your research work, it's amazing. All what I can say, that this 2 bytes CRC is somebody's misunderstanding. Really it is applicable only for FSK mode.

closed 2022-11-26 by Radiomix2000