tapparelj/gr-lora_sdr #35: Question about timing shift

lizhenqianri, 2022-12-13:
Hello, I have been following this project. I found that after adding the Channal Model module, setting some time offset can cause a wide range of bit errors. Excuse me, is there any solution?
[image]

tapparelj, 2022-12-15:
Hello @lizhenqianri,
- The documentation about the [channel model block](https://wiki.gnuradio.org/index.php/Channel_Model), the Frequency offset should be set as a normalized frequency.
- What oversampling factor do you use for the receiver? 
- For the epsilon, iirc the channel model uses a MMSE resampler which requires the input signal to already be over-sampled. Can you try producing samples out of the modulate block that are oversampled by a factor 4 before feeding it in the channel?

lizhenqianri, 2022-12-15:
thanks，@tapparelj 
The first time I found this problem was when using B210+BUC for long-distance communication, using SF=7 and 8, I found that it might be caused by time offset, so I used gnuradio to conduct a time offset simulation test (frequency offset is set to 0 ), the test results are consistent. When sending 200 bytes, the received data starts to get error at about 30 bytes.
[image]

tapparelj, 2022-12-15:
I see what is th issue:
- The estimation of the sampling frequency offset is based on the fact that most radio will use the same oscillator to generate both the sampling frequency and up-conversion signal. 
- This links both CFO and SFO. As we can estimate efficiently the CFO, we use that estimate to get the offset of the generating oscillator, in term recovering the SFO.
- If you set the CFO to zero, the estimate of the SFO will also be zero and won't be compensated. This leads to the frame slowly accumulating sampling time offset until the symbol recovered is off by one, leading to constant demodulation error for the rest of the frame.
- To simulate a real radio, you should define a clock frequency offset in gnuradio and deduce both CFO and SFO from it.

closed 2023-01-17 by tapparelj