tapparelj/gr-lora_sdr #36: Dobout about RFM95 Library

jp-polito, 2023-01-23:
Hi! Firstly i will congratulations for this work! I try to test lora RX with gnu Radio for receive data on my Sx1278 module, but i dont got sucess, im using one RTL_SDR for receive the datas. What library you use for test with RFM95? You have one exemple you used for this test? Thanks you so much!

tapparelj, 2023-01-24:
Hello @jp-polito,
I've just tested again with the RFM95 and here is what I use:
-  The library RadioHead (you can view the documentation and download it [here](http://www.airspayce.com/mikem/arduino/RadioHead/))
- I've attached a sample arduino code for the transmitter [basic_tx.zip](https://github.com/tapparelj/gr-lora_sdr/files/10487714/basic_tx.zip)

A few notes though:
- The library RadioHead adds 4 bytes in front of your preamble, for addressing and information flags.
- In my sample code I set them to 0x20 such that they will be printed as 4 spaces by the GNU Radio receiver
- For the other direction (USRP as Tx and SX1278 as Rx) you have to set properly the address  "To" for the node to print the message (e.g. by setting it to 255 for broadcast). You can set the RF95 headers by inserting the block named `Rh rf95 header` between the `message strobe` and `whitening block` in the example lora_TX
- The GNU Radio implementation works with any library able to set the SX12xx registers through the SPI link. So if this header prepending is useless to you, you can try any other libraries that doesn't add it.

Let me know if something is not clear, or if you encounter any issue!

jp-polito, 2023-02-01:
Hi @tapparelj Thanks the Help! I'm Trying test the library but I don't has sucess, im using a different frequency of you im using 434 Mhz and RTL-SDR instead USPR and im using a RFM98PW, but on datasheet is same of RFM95. I'm  send the .grc code and the arduino code. 

[ZIP.zip](https://github.com/tapparelj/gr-lora_sdr/files/10561673/ZIP.zip)

tapparelj, 2023-02-02:
Hello,
the sample rate of the RTLSDR source need to correspond to the "LoRa bandwidthi" * "os_factor".Therefore, you should set it to 500 MSps here. If your radio doesn't support this sampling rate, you will need to use a rational resampler (also available as a block).
Let me know if it fixes the issue.

closed 2023-02-25 by jp-polito