tapparelj/gr-lora_sdr #39: Help me to understand how to input bitstream

Radiomix2000, 2023-02-12:
I'm trying to input a bitstream (GR type Char/uChar) to LoRa graph, but cannot figure out, why it doesn't work.
I tried to convert Tagged Stream to PDU and send it to whitening. But get error:
```
Uflowgraph :warning: Could not find port msg in:
flowgraph :warning:   system
flowgraph :warning:   msg
Traceback (most recent call last):
  File "/home/armin/Gnuradio/OOT/gr-lora_sdr/examples/tx_rx_functionnality_check.py", line 241, in <module>
    main()
  File "/home/armin/Gnuradio/OOT/gr-lora_sdr/examples/tx_rx_functionnality_check.py", line 219, in main
    tb = top_block_cls()
  File "/home/armin/Gnuradio/OOT/gr-lora_sdr/examples/tx_rx_functionnality_check.py", line 108, in __init__
    self.msg_connect((self.pdu_tagged_stream_to_pdu_0, 'pdus'), (self.lora_sdr_whitening_0, 'msg'))
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 51, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 127, in msg_connect
    self.primitive_msg_connect(*args)
ValueError: invalid msg port in connect() or disconnect()
```
The whitening block has 2 types of input: Message Strobe and File Source.
I tried both with no success.
How is it possible to forward to LoRa graph, particularly to whitening block, a bitstream, or bitstream modified to some packet data or whatever appropriate for whitening block? Can you point, what data type conversions I have to do to make this possible?

This is a part of my graph, where I'm trying to send bitstream to LoRa.
[image]

tapparelj, 2023-02-13:
Hello,
The input of the whitening block can either be a message PMT or stream of bytes.
- The message PMT should be a polymorphic type contaning a string, for example pmt.intern("The string you want to send") .
- The file input should be a sequence of chars with frames separated by commas for example a file source containing "first packet,second packet," will generate two packets. The isHex parameter let you have comma separated string of hex values (e.g. "6669727374207061636b6574,7365636f6e64207061636b6574,").
I've just pushed a version that let you choose the separator you want to use for the File source input and fix a small bug for the isHex option. Don't forget to run `make uninstall`, `make clean` and then `make install` after pulling it.

Let me know if it is still confusing or if you encounter any other issue.

Stianje, 2023-03-06:
I tried to update, and install the newly pushed update.
It worked fine before, but now I get this error:
[image]

How to troubleshoot this? Thanks

Radiomix2000, 2023-03-15:
Hey hey! :) You can close it temporary, but I'm still making some experiments, so just not ready to report about them. But first thing which strike the eye, is unstable frame beginning detection. I used stream mux and injected "," ([0x2C]) symbol into stream to define packet beginning. But sometimes this hex combination appears in bitstream inside packet and becomes the cause of wrong packet beginning detection. I'm not sure how to overcome that. Maybe using Gnuradio build-in tagged stream capability can solve this? What if whitening block has option to set tag name, which designates packet beginning in bitstream? So we just using "Stream to Tagged Stream" block and have tagged stream before whitening block. Then whitening block gets stream and reads tag, identifying the packet beginning.

tapparelj, 2023-03-16:
Hello,
Sorry about closing the issue, I didn't realized it wasn't you who wrote the last message with the edit saying it was fixed. 
If I understand correctly, you use a `File Source` containing many bytes and insert the commas afterward? Then, if the bytes in your file can randomly have the value 0x2C that will indeed cause troubles. I could find a quick workaround, but it is obviously ugly, especially the lambda.
[image]
The code for the lambda is 
```lambda x: time.sleep(delay) or pmt.intern(str("").join([chr(x) for x in (pmt.u8vector_elements(pmt.cdr(x)))]))```
and it must be in "Raw Message" mode.

The solution with adding tags is obviously much better. However, I don't have the time to add it right now, but if you add the option to the whitening block (it should be easy by searching for a length tag, and if one is present, bypass the check for separators) I would be happy to do a merge!

Radiomix2000, 2023-03-19:
I'll try your solution. 
Trying to do a merge as well, but afraid, my c++ skills are so weak, that it will take a time to modify the code, adding length tag. Can I ask you some questions, during my work with code?

Radiomix2000, 2023-03-21:
Hi,
I made a graph as you suggested
[image]
and set the payload length = 48 bytes.
But when I received packets, the payload size was not fixed and not equal to desired value. As a result it was not possible to reconstruct payload structure on receiving side and not possible to get a correct position in bitstream, where the new voice frame starts from (I'm working with sending a codec2 voice frames though LoRa). Is it possible to set a fixed packet length? On RX side I can set it, and RX side works flawlessly with hardware transmitter, showing me correct frame structure and fixed payload length.

This is what I get in console on RX side. Payload length is not fixed.
```
--------Header--------
Payload length: 73
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: FÃL ±­¥O'Ißï!WQ»!ðkÃ[CÛØá?B°I@

CRC valid!

--------Header--------
Payload length: 69
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: ÀOó-`@G­8Òè4äÀÃM9âBÏ-#	GGU#ÜÈ9ÆÁÛ3§%f

CRC valid!

--------Header--------
Payload length: 69
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: CEr³Á
}`CñKACÃÎº´GÀCei6Y±ÄFÅ
av{=¤

CRC valid!

--------Header--------
Payload length: 75
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: Ø;ÒÜ¥¤ÀQßök¬ÅÇY#p­a
]Mv¼á@Ñ'2ª¥Áôû!P¨¤

CRC valid!

--------Header--------
Payload length: 76
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: |ØK¡ÃÁhÞ9rØ$fÐR­ìÀCãÂÈÌÂÀEmÂ÷ÌÃâB´;a

CRC valid!

--------Header--------
Payload length: 70
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: ÃC%KÅÍ%KÍõ¡AÃ*Q
ÔÁC,ÉÅV'ÀCm3ÔCBcE5 

CRC valid!

--------Header--------
Payload length: 75
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: @CÇ!©§ÁÃ¢Þ£AÇáb(eâGÈ)¢ÍÎø0AÌÞ©§

CRC valid!

******* MESSAGE DEBUG PRINT ********
(((rate_now . 401.435) (rate_avg . 395.153)))
************************************

--------Header--------
Payload length: 71
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: ÆÔúX9cðÞ^H9GÕE:Tø-ÁMÈÈ9]ë+PØ$æÃðÙkR®1Ç

CRC valid!

--------Header--------
Payload length: 78
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: pkÔ¬EùzÞ÷ä¬	XÙÓiÍ@Ã¯IÍåÇÀÃ
ñMçÝëK¼§

CRC valid!

--------Header--------
Payload length: 66
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: ÆPZÃÍçM%>5CÂEª-àÀO
(à@EaCÈ)fÀM ÊH;

CRC valid!

--------Header--------
Payload length: 73
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: ÁÃ£Lû@CÂÛü³%CC"ÓZêæÍËÓXÞE
CCÒï½¢EMÍ.Ø´)#

CRC valid!
```

tapparelj, 2023-03-22:
Hello,
Sorry, it seems it was a too dirty fix. It worked in my test case but obviously causes issue in yours.  
You will need to add properly the option to use length tag in the whitening block, else encoding the bytes in base64 and still use a defined separator would work if you don't mind the extra payload length, but in any case you would need to write some code.
Sorry not to have an easier solution.

tapparelj, 2023-08-14:
Hello,
I've just pushed a new version that supports tagged stream with payload length to separate the successive payload contents, alongside a few handful improvements.
I will therefore close this issue. 
Thanks again for pointing out that it could be a useful capability!