tapparelj/gr-lora_sdr #40: Error: "No module named 'gnuradio.blocks.blocks_python"

HaoHan9717, 2023-02-15:
Hi, a pretty nice implementation on GNUradio and Python. 

I have tested it on GNUradio successfully. But when I turned to Python on Ubuntu 20.04, I got the first error titled "no module named 'gnuradio' ". And then I follow the tips you gave us in the part of "Frequent Issues" and then it changes to another error named "No module named 'gnuradio.blocks.blocks_python' ". But I have already finished running all the commands you gave us in the "Frequent Issues" part. So I do not know how to fix it for now and hope to hear from your advice. 

Thank you!
[image]

tapparelj, 2023-02-15:
Hi @HaoHan9717,
If I understand correctly, it is working in gnuradio-companion (grc), but not when executed directly in the terminal, is it correct? If yes, you probably won't need to use the `export` commands to set the PYTHONPATH and LD_LIBRARY_PATH by hand.
Let's try to find what are the differences between grc and the terminal
- Did you have gnuradio installed only through the conda environment, or also via source or a packet manager?
- How do you launch grc? From a command in the terminal, or through the application icon?
  - If from the terminal, in which conda environment are launching it?
- When you run the flowgraph from grc can you tell me what command is actually run ? it should be something similar to `Executing: /home/{username}/conda/envs/gr310/bin/python -u /home/{pathToRepo}/gr-lora_sdr/examples/tx_rx_functionnality_check.py` showing up in the terminal window of grc (bottom left by default)

HaoHan9717, 2023-02-16:
Hi @tapparelj ,

Thanks for your early reply. Since I am new to GNUradio, I did not run the code in grc, but directly in the terminal instead. And I tried the `export ` commands but it did not work.

- I installed the gnuradio using the package manager. But it can be used by typing in 
 `gnuradio-companion` in both the base and conda environments.

- At first I launched the grc using application icons on Ubuntu. After your mention, I realized the problem and launched it after `conda activate gr310`. And then I found the `your-prefix` and `python-version` were different with that in the base environment. So I rectify the relevant codes in the `~\.profile` and then reboot the system. But after the reboot, I log into the system and got the error as you can see in the following picture.
[WechatIMG363.pdf](https://github.com/tapparelj/gr-lora_sdr/files/10755993/WechatIMG363.pdf)

- As I cannot log into the system, I haven't tried the flowgragh from grc. Thus I cannot tell you which command is running. I have to figure out and fix the problem above and then I can test it thereafter.

tapparelj, 2023-02-16:
Hi,
If you have multiple installations of gnuradio, this might require some care to use things properly. 
By default my blocks will try to install themselves in a conda environment if one is active. This is practical when you don't want or can't install them as sudo. 
- If you want to use the conda environment, you shouldn't have to set any "export" manually as the block should get installed in standards folders in your conda environment. The only requirement then is that if you want to use gnuradio-companion, you have to launch it from a terminal using that environment (with the command `gnuradio-companion`). When launching a script from the terminal, you should just have to use the correct python runtime, which should be the one of the conda environment.
It's very surprising that you managed get such an error by appending paths to PYTHONPATH and LD_LIBRARY_PATH, I have never seen that causing such an issue.
- If you prefer to not use conda at all, it is also entirely possible, you can deactivate (or remove the conda environment gr310) and install the blocks with `sudo make install`.

Let me know how it evolves.

HaoHan9717, 2023-02-20:
Hi, Sorry for my late reply.

After taking your advice into practice, the problem has been solved perfectly. It seems that I kept multiple versions of GNUradio and launched them both in the base and conda environments. 
And your solution is very helpful that by only launching GNUradio in the conda environment will the problem vanished entirely.

Thanks for your generous help.

closed 2023-02-20 by HaoHan9717