tapparelj/gr-lora_sdr #41: CRC Error when using USRP N210

HaoHan9717, 2023-02-21:
Hi, @tapparelj 

Recently I was using the GNUradio implementation of gr-lora_sdr on 2 USRP N210 devices (a pair of transceiver), which is nearly unmodified compared to your released version. However, I got several wrong `rx msg` (a string of gibberish) with the notification `CRC invalid`. But I noticed that my two devices are placed near to each other which should not cause any problems during the air-channel transmission.

Can you please tell me how I can fix it? Thank you very much.
[image]

tapparelj, 2023-02-21:
Hi @HaoHan9717,
Are you using the implicit header mode? If it is the case, the receiver needs to know in advance the length of the payload, the CRC presence, and the coding rate.
It look to me as if the receiver expect far more payload bytes than what you send. 
If it is not the case, can you set the "print_header" option of the `Header decoder` block to "yes"?

tapparelj, 2023-02-21:
Hi,
Indeed the packet error rate should be very close to 0 for a high SNR. I would suspect that something in your configuration is still not entirely correct. Could you send a screenshot of the flowgraph you use? I would like to have a look at your sampling rate, oversampling factor, etc..
In addition, is there any warnings during the initialization of the USRPs?

HaoHan9717, 2023-02-22:
Hi, thanks for your instant reply.
So the picture below is the configuration of the GNUradio flowgraph. Please check it out and put forward your comments.
[image]
And for your next consideration, I did see a warning during the initialization and it should be like this `The buffer could not be resized sufficiently, please run: sudo sysctl -w net.core.rmem_max=50000000`. Thus, I ran the commands which is suggested by GNUradio and I did not see that warning after I ran this command. Usually, the notifications are like the one below:
[image]

HaoHan9717, 2023-02-23:
Hi, @tapparelj 

Today I tried to switch the USRP transmitter with the one of receiver and it worked well with zero error rate. So I guess maybe my devices have some internal hardware errors which I did not notice before. Anyway, If you have any more comments on the parameters of the flowgraph above, please guide me on how to fix it. If not, this one works well for my new settings and no need to change it.

Thank you.

tapparelj, 2023-02-23:
Hi,
The only thing that comes to mind is that having only an oversampling of 2 might cause a rough compensation of the SFO, particularly if it is large between your two USRPs. I would try to increase that to 4, but aside I don't see anything else standing out.

closed 2023-02-23 by tapparelj