tapparelj/gr-lora_sdr #42: sync word length

fufellen, 2023-02-27:
can't find setting to set preamble and sync word length
Is it possible to do this?

tapparelj, 2023-02-28:
Hello @fufellen,
The Sync words can be set in the `modulate` block for the transmitter and `Frame sync` block for the receiver. They can be given as either a decimal value, a hex of directly the value of the sync words symbols (e.g 18, 0x12 or [8,16]). The sync words do not have a length parameter as they always consists in two modulated chirps in the preamble. Some devices use 2 bytes to set the sync words, I however don't have any at hand to test, but from what I've seen the relationship should be: for sync word 0xabcd : [x,y] =  [a<<(b-1), c<<(d-1)] . I'm however not sure what is happens if b or d are equal to 0.
The preamble length can be set in the same two blocks.
Let me know if something is confusing, or if you have the chance to verify this relashonship if your device uses the 2 bytes sync word format.

fufellen, 2023-03-13:
i can't find a setting to set preamble length,
maybe it is implicitly included in one of the block parameters?

can't accept package on sf 5 and sf 6
on some generations there is no backward compatibility for these factors
Can your project support different generations?
i use analog
sx 1262

tapparelj, 2023-03-13:
Hello,
the preamble length option was added in version v0.5.3. If you use an older version it is normal that you can't find this option. If you pull the new version, be sure to run `make uninstall` and`make clean` before `make install`.
I'm not aware of the exact differences introduced for the sx1261/2 and don't have one under the hand to test. The sf5 and 6 were tested with the SX1276.
Thanks for bringing that up, I'll add this information on the ReadMe

tapparelj, 2023-04-11:
Hello @fufellen,
you should be able to change the preamble length. For me there is no issue with SF 6 (cf. fig). 
Did you try to use the explicit header mode? Because it wasn't supported by previous generation of LoRa devices, therefore it isn't implemented in this repo yet.  
[image]

closed 2023-06-18 by tapparelj

nkaramolegos, 2023-10-31:
Hello, can you @tapparelj and @fufellen  please inform me how you create this beautiful captures' plots?

Also, I want to receive data from a device and I don't have information for the sync word how can I proceed? As I far I know the TX and RX should have the same sync word

tapparelj, 2023-10-31:
Hi,
On my side I just saved the output of the _modulate_ block with a _filesink_ and then opened it with matlab and plot the instantaneous frequency with something like plot(diff(unwrap(angle(tx_samples))))
For finding the network identifiers without having access to the transmitter configuration, you can print the values bin_idx that are checked as network ID at lines 581 and 593 of lib/frame_sync.cc. (I can't run any code now so maybe you need to investigate a bit by yourself)

fufellen, 2023-11-28:
> Hello, can you @tapparelj and @fufellen please inform me how you create this beautiful captures' plots?
> 
I was using something like (or maybe it) Adobe Audition. There is a spectrum view of .wav files.