tapparelj/gr-lora_sdr #43: An issue with not receiving or printing out the complete packet content in its entirety

Johnn13, 2023-03-02:
First of all, thank you very much for contributing this great work~~
I have a question  when I send packets using commercial LoRa nodes（based [RH_RF95](http://www.airspayce.com/mikem/arduino/RadioHead/classRH__RF95.html)）, I can only receive the first two bytes of the packet forever, and the next bytes cannot be received or printed
[image]
I had modified the USRP Source module parameter,such as Minoutbuf,Maxoutbuf,but that won't help.
My version:
GNU Radio 3.10
Ubuntu 22.04
the newest version gr-lora_sdr

tapparelj, 2023-03-02:
Hello @Johnn13,
Given that the frame passes the CRC check, it seems that you only send 6 bytes. As the radiohead library adds 4 bytes of addressing and flag in the start of a frame, are ou sending only two bytes? If yes, are those bytes supposed to be printable ASCII characters or have any value?
In this view, the received bytes are printed as ASCII characters and some are not printable (i.e. 0x00 to 0x1F and 0x7F).
If you want to recover the bytes values themselves, you have to use one of the output of the `CRC verif` block.

Johnn13, 2023-03-03:
hello, @tapparelj,thank you for your reply.
In my comercial lora node,the contents of the packet I sent are:`node1: IOT 20230302 hello word.`But it only outputs the first two bytes: `no`
In fact, I suspect that there is a problem with the calculation of the length of the Payload in the CRC block.However, I didn't find the payload length calculated in which section during the whole project when I used the explicit header module.
at the end,Could you please be specific about the meaning of your last sentence?`If you want to recover the bytes values themselves, you have to use one of the output of the CRC verif block.`
Looking forward to your reply~~~

tapparelj, 2023-03-03:
Hello,
The payload length is directly read from the explicit header which, in your case, only mention 6 bytes of payload. Since both the header checksum and the CRC are valid, I would check two things:
- Are the message received in their entirety suing another commercial node as receiver?
- Can you double check that in the transmitter code, that the buffer containing the  bytes to transmit is sufficiently large to contain your entire message, before passing it to the function starting the transmission?

Regarding my comment on the utilization of the outputs:
- If instead of transmitting a payload containing printable chars ,as you do know given the payload content, you wanted to send a payload containing as an example the bytes 0x01,0x02,0x02,0x01,.. nothing would get printed in the terminal. However, they would still get output through the output message port or output stream where you can use them however you need.

Johnn13, 2023-03-06:
hihi~ @tapparelj ,thank you for your patience.
Based on your advises, we found that there might be a problem with our lora node‘s registers that prevented the full packet from being sent.As expected, this issue was solved when we used the new lora node.
By the way, under the implicit header module, can the gr-lora-sdr calculated the payload length? if the answer is yes,in which section during the process?

tapparelj, 2023-03-06:
Hello,
I'm glad to hear you could fix it!
In implicit header mode you have to specify the payload length on the receiver side, similarly to commercial LoRa devices.

closed 2023-03-09 by Johnn13