tapparelj/gr-lora_sdr #47: RX Block Emit payload as binary as alternative to String?

Timvrakas, 2023-04-18:
What is the preferred way to process the payload output from the RX block (from the crc verification block)?
The payload block appears to produce a PDM "Symbol" type, which if I understand right is a C++ string under the hood. However, my payload is not a string, and so any attempts to use `pmt.symbol_to_string(msg)` on the messages causes a UTF-8 parsing error.
What is the right way to handle the emitted data in a packet form? Is there a way to have the RX block emit the payloads as a "blob" or vector of bytes, instead of a String?

tapparelj, 2023-04-18:
Hello @Timvrakas,
I'm surprised  by the error you get as the string generated and passed as a pmt symbol comes directly from the concatenation of uint8_t which should all be valid UTF-8 characters. What happens if you connect the output to the print input of a "Message Debug" block?
The "CRC verif"  can also output directly the bytes as a stream. Outputting vectors of bytes is tricky as the size of this vector might be different for each frame (in explicit header mode) but the output size has to be already known by the constructor and I'm not aware of a gnuradio block that allows variable output/input vector length.

Timvrakas, 2023-04-19:
That explains why a vector of bytes is not used, thanks.
When I use the "Message Debug" block, the message prints like this:
```
******* MESSAGE DEBUG PRINT ********
��Hello World
************************************
```
The invalid UTF is rendered as ??

However, when I try to process with a Python Block:
```
class print_hex(gr.basic_block):
    def __init__(self):
        gr.basic_block.__init__(
            self,
            name="print_hex",
            in_sig=None,
            out_sig=None)

        self.message_port_register_in(pmt.intern('msg_in'))
        self.set_msg_handler(pmt.intern('msg_in'), self.handle_msg)

    def handle_msg(self, msg):
        print(msg.is_symbol())
        print(msg)
```
I get the following:
```
True
thread_body_wrapper :error: ERROR thread[thread-per-block[14]: <block print_hex(12)>]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
```

I can't seem to interpret the symbol as anything other than an encoded string. I suspect if I write a C++ block, I could handle the string as binary data and won't have an issue, but I wonder if it's possible to do this in Python. Does the PMT "Blob" type also require a fixed length defined ahead of time?

Does the stream of bytes interface include tags to mark the start and/or end of a packet, and CRC validity?

Timvrakas, 2023-04-19:
I see the byte stream is tagged:
```
Tag Debug: 
Input Stream: 00
  Offset: 17  Source: n/a     Key: frame_info   Value: ((err . 0) (ldro_mode . 1) (crc . 1) (pay_len . 54) (cr . 4))
```
Does `err` include CRC errors?
I feel this may be the most "GNURadio" appropriate solution, rather than passing the payload as messages.

tapparelj, 2023-04-20:
Hi,
The tag debug block has no issue recovering the string, it prints ?? for ascii symbols that are not printable (but their value should be correct). I've never tried to recover a message generated by a c++ block directly in python so I don't know if this can cause your issue. 
Regarding the blob, I never used it but you should probably be able to quickly try it by changing the line with the "message_port_pub" to use a blob based on message_str.data() and message_str.length() instead of directly using the string.
The tags are currently only used for the input of the crc verif block and I should have disable their propagation as their offset won't be correct at the output. The "err" only indicates header errors. 
Based on what you say, I think the most practical implementation would be: 
- a stream of bytes with tags indicating beginning, number of bytes and CRC result. 
- a message output containing a dictionary with keys: payload_as_blob, payload_as_string, payload_length, and crc_valid.

Timvrakas, 2023-04-25:
I got this working using tags and the byte stream output. Here's my python block for future reference:

```
import numpy as np
from gnuradio import gr 
import pmt

class print_hex(gr.basic_block):
    def __init__(self):
        gr.basic_block.__init__(
            self,
            name="print_hex",
            in_sig=[np.byte],
            out_sig=None)

    def general_work(self, input_items, output_items):
        """example: multiply with constant"""
        tags = self.get_tags_in_window(0, 0, len(input_items[0]))
        for tag in tags:
            print(f'Tag Key: {tag.key}')
            print(f'Tag Value: {tag.value}')
            print(f'Tag Offset: {tag.offset}')
            print(f'Read offset: {self.nitems_read(0)}')
            pay_len_pmt = pmt.dict_ref(tag.value, pmt.intern("pay_len"), pmt.PMT_NIL)
            pay_len = pmt.to_python(pay_len_pmt)
            print(f'Payload Length: {pay_len}')
            print(f'Buffer Length: {len(input_items[0])}')
            print(input_items[0][(tag.offset-self.nitems_read(0)):(tag.offset-self.nitems_read(0)+pay_len)])
            self.consume(0,pay_len)
        return(0)
```

closed 2023-04-25 by Timvrakas