tapparelj/gr-lora_sdr #50: Running example with USRP B210

ktolonen, 2023-05-10:
I'm trying to run examples with Ettus USRP B210 board. There is no RX prints appearing on terminal when UHD used. Functional test and simulation are working perfectly. I'm newbie with these tools, and not sure if this is due the my tool chain and if there is something on example to adjust. I see warnings about bandwidth on start. Just reporting this if someone has similar issues and knows the solution already.

```
[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.HEAD-release
[INFO] [B200] Loading firmware image: /home/kari/miniconda3/envs/gr310/share/uhd/images/usrp_b200_fw.hex...
[INFO] [B200] Detected Device: B200
[INFO] [B200] Loading FPGA image: /home/kari/miniconda3/envs/gr310/share/uhd/images/usrp_b200_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 32.000000 MHz... 
[INFO] [B200] Actually got clock rate 32.000000 MHz.
[WARNING] [AD936X] The requested bandwidth 0.125 MHz is out of range (0.2 - 56 MHz).
The bandwidth has been forced to 0.2 MHz.
usrp_source :info: set_min_output_buffer on block 1 to 512
[WARNING] [AD936X] The requested bandwidth 0.125 MHz is out of range (0.2 - 56 MHz).
The bandwidth has been forced to 0.2 MHz.
modulate :info: set_min_output_buffer on block 5 to 10000000
multiply_const_cc :info: set_min_output_buffer on block 19 to 10000000
Press Enter to quit: 

```

ktolonen, 2023-05-10:
@tapparelj  thanks for quick reply!

Unfortunately setting samp_rate to 500kHz doesn't change the situation (os_factor is updating to 4 automatically).
I'm running example/tx_rx_uspr.grc. I changed both samp_rate_tx and samp_rate_rx,

[image]

````
[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.HEAD-release
[INFO] [B200] Detected Device: B200
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 32.000000 MHz... 
[INFO] [B200] Actually got clock rate 32.000000 MHz.
[WARNING] [AD936X] The requested bandwidth 0.125 MHz is out of range (0.2 - 56 MHz).
The bandwidth has been forced to 0.2 MHz.
usrp_source :info: set_min_output_buffer on block 1 to 512
[WARNING] [AD936X] The requested bandwidth 0.125 MHz is out of range (0.2 - 56 MHz).
The bandwidth has been forced to 0.2 MHz.
modulate :info: set_min_output_buffer on block 5 to 10000000
multiply_const_cc :info: set_min_output_buffer on block 19 to 10000000
Press Enter to quit: 
````
I can run examples/lora_TX.grc and examples/lora_RX.grc on separate USRP B210 boards, and problem is same although RX is already using "samp_rate" 500k. TX has different configuration and there is a warning.

````
[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.HEAD-release
[INFO] [B200] Detected Device: B210
[INFO] [B200] Loading FPGA image: /home/kari/miniconda3/envs/gr310/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 32.000000 MHz... 
[INFO] [B200] Actually got clock rate 32.000000 MHz.
[WARNING] [AD936X] The requested bandwidth 0.125 MHz is out of range (0.2 - 56 MHz).
The bandwidth has been forced to 0.2 MHz.
modulate :info: set_min_output_buffer on block 4 to 10000000
Press Enter to quit: 
````
I'm using examples as those are on master. Only change is to remove hardcoded ip address.

tapparelj, 2023-05-11:
I don't have a b200 or b210 under the hand, but I tested it again on a b200 mini.
By just changing the sample rate in tx_rx_usrp.grc, I receive the packets without any issue. The terminal output is the same as yours, except for the device name, the GNU c++ version (11.2.0),  the UHD version (4.1.0.5-3), and I get the received messages printed. The only way I managed to not receive anything is by disconnecting the antennas. 
If you use only one radio, can you try setting the receiver to the RX2 port (in the "uhd: usrp source") and be sure that the antennas you use are suited for a carrier frequency around 868MHz? 
I wouldn't expect the version difference of UHD and C++ to make much difference. How did you installed gnuradio (from source, apt, conda, pybombs, or other) ?

ktolonen, 2023-05-11:
Then I suspect antennas since not known history for those and there is no markings. Changing receiver port or using two different boards doesn't change the situation. I have to source known good antennas from somewhere. I let you know when have been able to test with those.

I'm using Ubuntu 22.04 and installed gnuradio 3.10.3.0 (Python 3.10.5) from apt.

tapparelj, 2023-05-11:
A quick test using a cable to connect both ports should do the trick to identify if it is the issue.
I think the issue might come from the version of gnuradio. I tried to get the newest version 3.10.5 and this led to a runtime error. Something changed between the version 3.10.1 that I use and the newer ones and something seems to be broken. I will try fixing it as soon as possible, but in the meantime you probably have to revert to a previous version of gnuradio to use this module.
if you installed gnuradio using:
`sudo add-apt-repository ppa:gnuradio/gnuradio-releases` and then `sudo apt install gnuradio`, you can go back to the default version for ubuntu 22.04 by doing:
`sudo add-apt-repository ppa:gnuradio/gnuradio-releases --remove`
`sudo apt update`
`sudo apt remove gnuradio`
`sudo apt install gnuradio`
Else, you can try directly specifying the version you want to install as `sudo apt install gnuradio=3.10.1.1-2`, but i'm not sure it will work as easily.

ktolonen, 2023-05-16:
I got another set of antennas and the example starts working as expected. I could even revert the sample rate change, and hello world frames were still received. 

So confirming that the issue was with antennas not supporting low frequencies. Thanks for your valuable feedback helping out to solve the problem with my setup. Closing this.

closed 2023-05-16 by ktolonen

refealwang, 2023-06-14:
Hello this gentleman, I have the same problem as you.
When using two B210s, no matter how I change their antennas (4 antennas in the lab, two VERT2450: 2.4-2.48GHz, 4.9-5.9GHz; VERT900: 824-960MHz, 1710-1990MHz; and one flat antenna, which I guess comes with the B210), I end up with the same problem as you exactly the same problem.
But if I use a cable connection between the two B210s, this does not cause these kinds of problems (receiving Hello World normally).
So I'm very interested in your antenna replacement, can you tell me the type of antenna you replaced and what else you did?

ktolonen, 2023-06-27:
Unfortunately I'm not able to tell any characteristics about antennas. I found some from lab and the one pair works. No additional changes done, just replacing the antennas.

refealwang, 2023-07-19:
Thank you for your response. I have since modified the gain of the signal transmission to enable wireless communication between the two B210s. The default gain is 0.