tapparelj/gr-lora_sdr #52: Can I get the symbol sequence？

XieyangSun, 2023-05-21:
Before the USRP transmits, I want to understand what is the symbol corresponding to each packet? For example, the "Hello World", After the 11 characters are whitening, interleaving, etc., how can the converted symbols be obtained? 
In the lora_TX grc, I try to use File Sink to intercept and save the data after 'Gray demapping' and 'Modulate', but I only get some complex signals.
In fact, I am trying to understand the possibility of reverse LoRa at the level of hexadecimal symbols. gr-lora_sdr helped me understand lora and provided a great platform. Many thanks to the author~

tapparelj, 2023-05-22:
Hello @XieyangSun,
You should be able to use the "File Sink" to see all the intermediate values. The "File Sink" block will just write binary files, therefore you should tell the program you use to read them how to interpret the file content.
As an example in MATLAB, you can use
```
f = fopen(['file_sink.bin']);
values = fread(f,'float32'); % other values of interest in different parts of lora_TX would be int32 or uint8
% Complex values are stored as alternating real and imaginary part, which can be put back to complex with
% values_complex = values(1:2:end)+1j*values(2:2:end);
fclose(f);
```
I hope this help you go in the right direction, let me know if you have other issues or questions!

XieyangSun, 2023-05-22:
Thanks for your reply! I re-observed the implementation of modulate_impl.cc, and I noticed that Preamble and other lora header parts were added to the front of a packet in the Modulate block. (I used to think that the all symbols of the entire data packet were calculated and then mapped to the baseband signal. Now I realize this is not feasible due to the presence of 2.25 down-chirps...) What I can't determine is whether the int data stream saved by File Sink after gray_demapping is the decimal symbol of the original information?

For example, I'm trying to transmit 'hello-world',
[image]
Convert data-int.bin in my matlab to get 44 decimal numbers as:
[image]
It seems to be correct, because the maximum value of these decimal numbers does not exceed 2^7=128.

tapparelj, 2023-05-23:
I'm not sure about what you mean by "original information", the input values of the _Modulate_ block are directly use to modulate the LoRa chirps. The decimal values you observe contains all the content of the frame (i.e header, payload, CRC).

XieyangSun, 2023-05-24:
Ok, that's exactly what I was looking for. Thanks for the reply~

tapparelj, 2023-05-24:
I'm glad I could help! I'm closing this issue then.

closed 2023-05-24 by tapparelj