tapparelj/gr-lora_sdr #53: Print_rx_msg doesn't work

coaccac, 2023-05-30:
[image]

Thanks to your code, I can conduct various experiments related to LoRa. However, when I execute the tx_rx_functionnality_check, the content of rx_msg does not appear like attahced figure. There have been no code modifications. How can I resolve this issue? Thank you.

tapparelj, 2023-05-30:
Hello @coaccac,
I think this error occurs when nothing is connected to the output of the _crc_verif_ block (adding a _null_sink_ in gnuradio-companion should fix it).
It should have been fixed in a recent commit. Did you pulled and rebuild the repository recently?

coaccac, 2023-05-31:
The issue was resolved by inserting a null sink. It seems that the problem occurred because I kept using the previous version without using the latest one. Thank you for your response! :)

closed 2023-05-31 by coaccac