tapparelj/gr-lora_sdr #55: sched: <block frame_sync (8)> is requesting more input data  than we can provide

itay-grudev, 2023-07-27:
I am getting the following error when using a spreading factor of 12. Can you give me advice for further debugging?

[image]

```
block_executor :error: sched: <block frame_sync (8)> is requesting more input data  than we can provide.  ninput_items_required = 32784  max_possible_items_available = 8191  If this is a filter, consider reducing the number of taps.
```

tapparelj, 2023-07-31:
Hi @itay-grudev ,
This error indicates that the input buffer of the of the _frame_sync_ block is not large enough. This buffer size can be specified in the preceding block which is your _Soapy LimeSDR Source_.  
Under the "Advanced" tab of the _Soapy LimeSDR Source_ set the "Minoutbuf" to a value above 32784 (e.g. 65536 should fix this).
Let me know if it solves the problem

itay-grudev, 2023-07-31:
Thanks a lot.

closed 2023-07-31 by itay-grudev