tapparelj/gr-lora_sdr #57: eplaced USRP block with RTL SDR

sajadmathe, 2023-08-20:
Hi
I am a new in gnuradio.
I replaced USRP block with RTL SDR block and when i run lora_RX.grc flowgraph in GNURadio i see this outpot:

[image]

I don't understand anything from this output. Can you guide me?

thanx

tapparelj, 2023-08-21:
Hi @sajadmathe,
Based on the error, it seems that the SDR you are using doesn't support a sample rate of 125kHz. To know which rates it supports, try searching in its datasheets.
If it supports any multiple of 125kHz, you can simply add a _Keep one in N_ block to decimate the signal. Else you might need to use a _Fractional resampler_ or a _Rational resampler_.

In addition, I see some errors with the parameters you set in the block:
- The _Sample rate_ should be set to the variable _samp_rate_ (you can use variables in flowgraphs). The example uses an oversampling factor of 4 (that I would recommend to keep) to synchronize, hence being set to 500kHz for a LoRa bandwidth (_bw_) of 125kHz. 
- The _Frequency_ should be set to the variable _center_freq_. This variable of course needs to match the frequency used by your transmitter.

If you are new to gnuradio, try having a look at the tutorials [here](https://wiki.gnuradio.org/index.php?title=Tutorials) to get familiar with the main concepts of flowgraphs.

tapparelj, 2023-10-24:
I'll close this issue as there has been no follow-up in the past two months, but feel free to reopen it if needed.

closed 2023-10-24 by tapparelj