tapparelj/gr-lora_sdr #61: About collision scenario

coaccac, 2023-09-11:
I am using the provided LoRa implementation effectively. However, I have one question that may be related to the fundamental operation principles of GNURadio or LoRa. If a collision occurs between packets, can only the first incoming packet be decoded from the receiver's perspective? I am curious if the process of detecting the preamble of the subsequent packet only begins after the decoding of the previous packet is complete.
Thank you so much for tour great work!

coaccac, 2023-09-11:
To add further clarification, in essence, my question can be summarized as follows: Does the frame_sync process continue to search for the preamble even while decoding another packet?

tapparelj, 2023-09-11:
Hello @coaccac,
The current implementation of the frame synchronization only supports one frame at a time for a given spreading factor and carrier frequency. If a collision occurs, we will simply continue processing the first one and the demodulation success will depend on multiple factors such as their difference of power, carrier frequency offset, sampling time offset,...

coaccac, 2023-09-11:
If it were to operate as I've inquired, would it require significant modifications? Thank you very much for your response.

tapparelj, 2023-09-11:
Adding the capability to continuously evaluate if a new preamble is present should be quite easy. You would then need to instantiate a new receiver chain to handle the newly found frame. Creating those receivers on the fly is possible, but requires writing some code to manage the flowgraph during runtime. Else, you could just assume there would be a maximum number of colliding frames and have multiple receiver chains ready.
However, doing this simple changes would only work if the interfering frame has a substantially larger received power than the first one, and the first frame would still be lost. If the second frame has a lower received power than the first, you won't detect its preamble as we are looking for symbol repetitions to identify preambles.
If you want to recover the data of multiple colliding frames you would need to change the receiver quite a bit. If you are interested, you can have a look at those papers ([this](https://ieeexplore.ieee.org/abstract/document/9810018) or [this](https://ieeexplore.ieee.org/abstract/document/9644605)) which introduce two possible multi-users receivers as well as the issue of preamble detection. (if you don't have access to IEEE Xplore I can send them to you directly)

closed 2023-09-21 by coaccac