tapparelj/gr-lora_sdr #62: Compatibility of SF 5,6 with sx1261/2

jduses, 2023-10-25:
Hi,
Can you please tell why the SF 5, 6 are not compatible with SX1261/2 chips? Is there a different whitening sequence that they use or am I missing something?

tapparelj, 2023-10-25:
Hi @jduses,
for chips other than the SX1261/2, the datasheets indicates that the explicit header mode can't be used for SF<7.
This makes sense because the explicit header consists in 20 bits, which need to fit within the first 8 LoRa symbols sent with CR=4/8 and low datarate optimization, which exactly gives 20 bits for SF7.
To enable the explicit header for SF 5 and 6, they may have done many things (e.g. defining a different header structure, sent more symbols with CR 4/8 and LDRO, disable LDRO,...). I however didn't had the time to reverse engineer this for those chips yet.

tapparelj, 2023-10-31:
Hi @nkaramolegos,
based on the datasheets, there will be the same issue for sf5 and 6. I'll add this to the readme

nkaramolegos, 2023-10-31:
Can you send me the reference? Thanks

tapparelj, 2023-10-31:
Sure, it's described at page 36 of https://semtech.my.salesforce.com/sfc/p/E0000000JelG/a/3n000000qOUu/tpcW.qidS7dE9fxMVN_v5okCY64Xiq5OUeNNFal1a8c

nkaramolegos, 2023-10-31:
Do you have any hints on how to reverse engineer these chips?

tapparelj, 2023-10-31:
You can check [this report](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://www.epfl.ch/labs/tcl/wp-content/uploads/2020/02/Reverse_Eng_Report.pdf&ved=2ahUKEwiC9dz-4qCCAxUXEVkFHaMBBS0QgaEJegQINRAA&usg=AOvVaw2bv7oX1Yu61M5fj7kKaNXG)  on the reverse engineering of the rest of the physical layer to give you some hints on how you can proceed. And [this nice presentation](https://youtu.be/-YNMRZC6v1s?si=gUwmYm4OcobmpKyx).
I would be happy to accept a pull request if you add this capability!

tapparelj, 2023-11-04:
Hi, 
as there is no real question pending I will close this issue for now, but don't hesitate to reopen it if needed

closed 2023-11-04 by tapparelj