tapparelj/gr-lora_sdr #65: How to find the  location of 8 preambles and save them in frame_sync_impl.cc ?

sadsnow, 2023-11-14:
I want to completely extract and save the 8 original preamble symbols on the receiving side to the .dat file, but I have been looking for it in the frame_sync_impl.cc file for a long time, and I am not sure the location to process it.

I initially found that the relevant variables are as follows：

preamble_upchirps.resize(m_preamb_len * m_number_of_bins);
preamble_raw_up.resize((m_preamb_len + 3) * m_samples_per_symbol);

sadsnow, 2023-11-20:
@tapparelj @
Hello, and thank you very much for your response.
I have tried running the code at the two locations you mentioned and noticed that the output dimension is 128*8. Is this the dimension after down sampling?

This dimension does not align with my expectation for the 8 original preamble symbols. I am looking for the original preamble signal, calculated based on a sampling rate of 1Mbps, with a spread factor (sf) of 7 and a bandwidth (bw) of 125kHz. The expected signal dimension, calculated according to the sampling rate and symbol transmission time, should be 8*1024=8192. In the code, this dimension is described as `preamble_len * samples_per_symbol = 8*1024=8192`.

In essence, in addition to understanding the implementation and data collection locations of STO, CFO, and SFO, I am particularly interested in locating and extracting the raw data of the original preamble. How can I determine the exact positions of the 8 preamble symbols and extract this set of data?

Thank you again for your guidance on this matter.  I look forward to your response at your earliest convenience.

`
std::rotate(corr_preamb.begin(), corr_preamb.begin() + mod(m_cfo_int, m_number_of_bins), corr_preamb.end());

std::cout<<"dim of corr_preamb:"<<(m_n_up_req + additional_upchirps) * m_number_of_bins<<std::endl;
std::cout<<"m_os_factor:"<<m_os_factor<<std::endl; // also confused about m_os_factor and no answer
std::cout<<"preamble_raw_size"<<(m_preamb_len) * m_samples_per_symbol<<std::endl; // 8*1024

std::ofstream fout("/home/user4/Desktop/preamble.txt", std::ios::out|std::ios::app);
for(int i =0;i<(m_n_up_req + additional_upchirps) * m_number_of_bins;i++) 
{                                                                         
	fout<<"this is i"<<i<<":  "<<corr_preamb[i]<<std::endl;
}
fout.close(); 
`

> Hello @sadsnow, when you say original preamble symbols, you mean the ones without sampling time offset (STO) but still with carrier frequency offset (CFO) and sampling frequency offsets (SFO)? Or the ones after all offsets estimation and compensation? If you want the first case, I think that the easiest place to extract it is after line 664 from the _corr_preamb_. Else, it would be after line 673. (I haven't tried, only looked at the code. So don't hesitate to comment if it doesn't work as expected.)

tapparelj, 2023-11-21:
Hello,
so if you want to get the samples before decimation, you should get them from _preamble_raw_up_ before line 662.
The upchirps samples will be in that vector, but the STO is not corrected yet (i.e. the first sample of the vector is not the first sample of the preamble).
If you want to realign this vector you need to realign it yourself following a similar procedure as the one performed on the downsampled _preamb_correct_ in line 662 and 664.

closed 2024-01-18 by tapparelj