tapparelj/gr-lora_sdr #66: ModuleNotFoundError: No module named 'gnuradio.lora_sdr'

luoposhanCuiDongshan, 2023-11-22:
When I install gr-lora according to your steps, No matter I run the script file for the final detection or use gnuradio to build the flow chart according to your steps, an error will be reported in the end, and the error is always No module named 'gnuradio.lora_sdr'. May I ask what went wrong? How to solve it

tapparelj, 2023-11-23:
Hello @luoposhanCuiDongshan,
first I would strongly recommend to use the version for gnuradio 3.10 if you can, as there has been many improvements compared to the older unmaintained versions.
This error occurs when python doesn't find the module. There can be two main reasons:
- There was an error when building and installing the module
- The paths to the module is not properly setup in the variables used by python

For the first, did any error occurred during the `cmake ..`, `make`, or `make install`? If yes can you paste the building logs here?
If not this is probably the second reason and this should be fixable.
- How did you install gnuradio, through conda, pybombs, source,... ?
- Was a conda environment active in the terminal when you ran the installation?
- Did you install the module as sudo or with a prefix?
- Where did you installed the module? if you don't know, you can find it written in the terminal when you run `make install` in a line similar to 
```
-- Installing: /home/<username>/<install_prefix>/lib/libgnuradio-lora_sdr.so.v1.0-compat-xxx-xunknown 
```

luoposhanCuiDongshan, 2023-11-23:
> Hello @luoposhanCuiDongshan, first I would strongly recommend to use the version for gnuradio 3.10 if you can, as there has been many improvements compared to the older unmaintained versions. This [...]
I installed gnuradio3.8.0.0 using the following program
  git clone https://github.com/gnuradio/gnuradio.git
  cd gnuradio/ 
  git checkout v3.8.0.0
  mkdir build 
  cd build 
  cmake ../ 
  make
  make test 
  sudo make install 
  sudo ldconfig 

The gr-lora I installed was installed through the program you gave me. No errors were reported during the process. It was just that when I ran the cmake program, I did not follow the program you gave me. I only ran cmake ..

Another question is, in what folder should the command you gave be run under the terminal?
-- Installing: /home/<username>/<install_prefix>/lib/libgnuradio-lora_sdr.so.v1.0-compat-xxx-xunknown

If you find a file similar to gr310, what should be filled in xxx?
install /home/wk/miniconda3/envs/gr310/lib/libgnuradio-lora_sdr.so.v1.0-compat-xxx-xunknown

tapparelj, 2023-11-23:
Hello,
I'm a bit confused about how you want to install things. Can you answer to those questions such that I get an idea of how to help you setup things properly?
- Since you installed gnuradio from source is there a reason you want to use gnuradio 3.8 instead of 3.10 ?
- Do you want to install things globally or within a conda environment? 
- If you don't want to use conda, do you want to install it using sudo or not?
- Did you create a conda environment gr310 using the environment.yml file provided in this repo? 

The line "-- Installing: / ...."  was not a command that you should run, but an example of the line printed in the terminal when running `make install` (when you install the module from this repo) that allow to see where the module gets installed on your machine.

luoposhanCuiDongshan, 2023-11-24:
Because I use ubuntu18.04, the installed dependencies do not support gnuradio3.10.
If I use ubuntu22.04.3, then my uhd4.0.0.0 version cannot be installed.
I can't find the dependencies to install gnuradio3.10 version in ubuntu18.04, so I can only use gnuradio3.8
What is the difference whether it is installed globally? I just need to be able to use the lora module.
Regarding your last question, is it related to the following line of code?
conda activate gr310

tapparelj, 2023-11-24:
Thanks, it is more clear now.
First, the conda environment gr310 is used to install gnuradio 3.10 so you shouldn't use it. 
Then, since you installed gnuradio from source, directly as sudo the easiest is to just build this module in the same way.
By globally I mean by using sudo and installing it directly in /usr/lib which is the default location, contrarily as installing it in a folder in your home directory that would require you to manually configure Python to look in the folder you chose.
As it seems that you have conda installed, it might create problems by hijacking the installation path. 
So I would advise to:
- In the file gr-lora_sdr/CMakeLists.txt, delete the lines 35 to 38 to prevent conda to set the install destination to itself
- Go in the folder gr-lora_sdr/build (that you created when trying to install) and run 
```make uninstall```
- Delete everything inside the build folder
- Install again with:
 ```cmake ..```
```make -j4``` (the 4 is to use 4 cores in parrallel to speedup the build process)
```sudo make install``` (here you should see that the installation path is now something similar to /usr/lib instead of your conda environment)
```sudo ldconfig```
Let me know how it goes

luoposhanCuiDongshan, 2023-11-26:
When I do cmake.. I get the following error
CMake Error at CMakeLists.txt:38 (find_package):
   Could not find a configuration file for package "Gnuradio" that is
   compatible with requested version "3.10".

   The following configuration files were considered but not accepted:

     /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.5.git

-- Configuring incomplete, errors occurred!
See also "/home/wk/workarea/gr-lora_sdr/build/CMakeFiles/CMakeOutput.log".

If I want to install lora that is adapted to this gnuradio version, how should I do it? Why is it that when I install lora that is adapted to gnuradio 3.8 according to the branch you gave me, the output is still gr310?
https://github.com/tapparelj/gr-lora_sdr/tree/unmaintained/gnuradio-3.8_v0.3

tapparelj, 2023-11-27:
Are you sure that after you cloned the repository, you switched to the correct branch? You can check on which branch you are on with `git status`.
in the 3.8 branch, the CMakeLists.txt clearly states:
l93: "find_package(Gnuradio "3.8" REQUIRED COMPONENTS blocks analog filter fft)".
It is in the **main branch** that the CMakeLists.txt looks for gnuradio 3.10
l43: "find_package(Gnuradio "3.10" REQUIRED)"

I don't understand why you refer to gr310. gr310 is a conda environment never mentioned or used on the unmaintained/gnuradio-3.8 branch.

tapparelj, 2024-01-18:
As it has been some time without follow-up, I'll close the issue. Don't hesitate to reopen it if needed.

closed 2024-01-18 by tapparelj

JYK2001, 2024-12-02:
Hello, I'm having the same issue now, referring to what you mentioned earlier on how to install into gnuradio3.8, resume oot module and debug, and finally found this error when running gnu