tapparelj/gr-lora_sdr #67: Error in receiving and decoding msgs with SF 5 and SF 6

jds0987, 2023-12-06:
I am using heltec automation LoRa chip for transmitting LoRa packets. Using the LoRa Rx flowgraph in examples, I am not able to rx msgs for SF 5 and 6 . SF 7 to 12 works absolutely fine. Screenshots att
[image]

[image]

Kindly help.

tapparelj, 2023-12-06:
Hello @jds0987,
currently, this repo doesn't support the explicit header mode for the spreading factor 5 and 6 used by the SX1261/2/8. 
If your device uses one of those chips then it's expected that it won't work (you can have a look at #62 that also concerned that)

jds0987, 2023-12-06:
Can this block be modified for Sx1261/2/8 for SF 5 and SF 6? 
Will share the recorded files with SF 5 and SF 6 if required....

tapparelj, 2023-12-06:
I'm planning to start working to add support for those spreading factors as the next improvement to this repo. If the reverse engineering goes well, I should be able to add it to the repo before the end of the year.
I'll let you know when I have something working.

jds0987, 2023-12-07:
Ok....pls let me know if any assistance is required... 

Also I am trying to use rtlsdr as source in the rx flowgraph, but the flowgraph keeps throwing error 

No device detected, however all the libs pertaining to RTL have been installed.

Y is it so?

tapparelj, 2023-12-08:
Hi,
thanks for the test! What is the exact model of your LoRa node? I've noticed that some Heltec nodes uses the sx1276 and not the sx1262 (the one I've tested on my side).
If you indeed are using an sx1262 (it should be written on the chip), and you have the time, would it be possible to run the code with the version I just pushed on the feature/sf5_6_sx1276 branch with the following parameters:
- set the receiver variable soft_decoding to False
- Transmit with:
  - SF5
  - CR1
  - the payload "Hello World!"
  - Explicit header
  - CRC on
  The raw symbol values will get print in the terminal and that will be helpful to compare with what I get on my side.

jds0987, 2023-12-08:
Made the changes as u told earlier, however not decoding with SF5 and 6 . Sharing screen shots of flowgraph and output.

[image]

Flowgraph
[image]

jds0987, 2023-12-08:
This myt be the header, that is getting printed. I ran the test for SF7 it is giving the following output.

[image]

Probably, the flowgraph initiates by comparing the header values, Post comparison it starts decoding. For SF7 the header comparison passes and the transmitted msg is then decoded. 

But for SF 5 the comparison process is not getting through and the header checksum fails. Probably The header checksum block for SF5 and SF6 need to be modified.

[image]

jds0987, 2023-12-08:
Yea Sure........Here it is

[image]

Hope..... this is what u required......Pls let me know if anything else is reqd.

tapparelj, 2023-12-08:
This is intriguing, if you have those values as well then the header decoding should be handled exactly the same as for me.
Can you pull and install the version I pushed now on the branch?
Just to be sure, to pull and build the code on the new branch, you did
```bash
git clone https://github.com/tapparelj/gr-lora_sdr.git
git checkout feature/sf5_6_sx126x
cd gr-lora_sdr
mkdir build
cd build
cmake ..
make -j4 (the 4 specify that you can use 4 cores to build instead of the default 1, you can change to whatever fits your system)
sudo make install
```
This new version should work for sf5 and print the values if you set the soft-decoding to false. For me with sf5, the terminal prints
```verbatim
0x7
0xc
0x1f
0xc
0x6
0x7
0x1
0x6

--------Header--------
Payload length: 12
CRC presence:   1
Coding rate:    1
Header checksum valid!

0xa
0x1d
0xf
0x11
0xb
0x16
0x7
0x1c
0x1a
0x7
0x9
0xf
0x1e
0xd
0x3
0x14
0x14
0x9
0x9
0x12
0x16
0x1b
0x1a
0x18
0x1b
0x1e
0x3
0x19
0xd
0x1d
rx msg: Hello World!

CRC valid!
```

jds0987, 2023-12-08:
Hi,
The branch has worked. I was taking samp_rate to be 2e6, however when i used 500KHz, the flowgraph started decoding but it still misses out on few instances and give Header Checksum Invalid. Can we troubleshoot the same. Sharing a snippet of the same as follows.

```
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 2426666 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=2426666
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 2426666 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=2426666
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
[WARNING] [UDP] The current send_buff_size of 212992 is less than the minimum recommended size of 307200 and may result in dropped packets on some NICs
[WARNING] [RFNOC::GRAPH] One or more blocks timed out during flush!
usrp_source :info: set_min_output_buffer on block 1 to 136
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 2426666 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=2426666
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 2426666 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=2426666
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 307200 and may result in dropped packets on some NICs
[WARNING] [UDP] The current send_buff_size of 212992 is less than the minimum recommended size of 307200 and may result in dropped packets on some NICs
Press Enter to quit: D0x1c
0x10
0x6
0xc
0x1b
0x17
0x14
0x19

--------Header--------
Payload length: 41
CRC presence:   0
Coding rate:    2
Header checksum invalid!

0x7
0xc
0x1f
0xc
0x6
0x7
0x1
0x6
--------Header--------
Payload length: 12
CRC presence:   1
Coding rate:    1
Header checksum valid!

0xa
0x1d
0xf
0x11
0xb
0x16
0x7
0x1c
0x1a
0x7
0x9
0x8
0x1e
0xd
0x1c
0x14
0x14
0x9
0x9
0x12
0x16
0x14
0x1d
0x1b
0x1a
0x1a
0x1
0x9
0xd
0x19
rx msg: Hello world!

CRC valid!

```

jds0987, 2023-12-11:
Hello @tapparelj 
Can this code be made functional for GNURadio 3.7.13.4 as well?

tapparelj, 2023-12-11:
There is no reason I can think of that would prevent it. Initially, it was developped on GNURadio 3.7. The core c++ files (the ones in /lib ) should work fine, probably the asynchronous message passing callback should be changed back to the form they had before (just the way you declare the callback changed). 
> Hello @tapparelj Can this code be made functional for GNURadio 3.7.13.4 as well?

jds0987, 2023-12-12:
Hi @tapparelj 

Was playing around with this flowgraph and tried to test the demodulation and decoding of few of the lab recordings I had with me using File Source in thelora_RX flowgraph for SF5.

I have been getting this
Output

"CRC not supported for payload smaller than 2 bytes."(Screenshot attached).

[image]

Why is this happening and how can we troubleshoot this?

tapparelj, 2023-12-12:
Are you sending a payload of less than 2 bytes?
The CRC for payloads of less than two bytes is not supported yet as the last two bytes of the payload are considered in a special way for the CRC computation and I never had the time to check what they do when the Payload length is <= 2.

jds0987, 2023-12-13:
Hello @tapparelj 

Yes, The payload is less than 2 bytes. Are you planning to add support for payload less than 2 bytes?

jds0987, 2023-12-13:
> Are you sending a payload of less than 2 bytes? The CRC for payloads of less than two bytes is not supported yet as the last two bytes of the payload are considered in a special way for the CRC computation and I never had the time to check what they do when the Payload length is <= 2.

Hello @tapparelj

Yes, The payload is less than 2 bytes. Are you planning to add support for payload less than 2 bytes?

tapparelj, 2023-12-13:
Hi @jds0987,
yes, it's planned, but probably only around February or March.

jithivk, 2023-12-14:
@tapparelj Why the received data not showing in its hex format, rpp version
is doing that well.

jds0987, 2023-12-14:
Ok......Looking fwd towards this...... Wish u all the best.

tapparelj, 2023-12-14:
> @tapparelj Why the received data not showing in its hex format, rpp version is doing that well.
You would like to print in the terminal the hex content? For example "0x68 0x65 0x6C 0x6C 0x6F" instead of "hello" ?
That's a good idea and a simple thing to add. I'll add it very quickly. 
I'll keep you updated.

jds0987, 2023-12-14:
Hi @tapparelj 

Can the code be modified to get the header also printed for each spreading factor?

tapparelj, 2023-12-14:
> Can the code be modified to get the header also printed for each spreading factor?

I'm not sure to get what you mean by that. The explicit header content should already be printed for any spreading factor if you set the parameter _print_header_ to true in the _header decoder_ block

tapparelj, 2023-12-14:
> @tapparelj Why the received data not showing in its hex format, rpp version is doing that well.

Hello @jithivk,
I've just pushed an udate to the main branch that allows to select the format of the printed payload values in the terminal.

jithivk, 2023-12-15:
@tapparelj ,some LoRa IQ file make the application force shutdown or hang why it is happening like that.

jithivk, 2023-12-15:
what if i can record the active LoRa signal saved in its IQ format, that will help me further analysis in the future, auto detection and recording if possible

tapparelj, 2023-12-18:
@jithivk and @jds0987,
if your question has no link with the original topic of this issue, please open another one to keep things easier to follow an discuss.

jds0987, 2024-01-18:
@tapparelj 

Hi

Can u please modify the sf5 and sf6 branch to give hex output

tapparelj, 2024-01-24:
Hi @jds0987,
I pushed a new version on the branch where you can select between the ASCII and HEX output to display in the terminal.

jds0987, 2024-01-26:
Hi @tapparelj 
Thanks for the new version, it works like a charm👍

atambi802, 2024-07-17:
@tapparelj Are you planing on merging it to master? Thanks again!

jithivk, 2024-07-18:
Can u make the code to display single byte data support

MrTalon63, 2025-02-07:
Hi @tapparelj,
I hope you're doing well. I'm checking in regarding the SX126X implementation. Specifically, has compatibility with SF5/6 been merged into the master branch yet? We're planning to use this implementation in our upcoming CubeSat ground station project, so any update you can provide would be greatly appreciated.

Andrea99-beep, 2026-11-03:
Hi @tapparelj, can I ask if the master branch now is able to decode sf5 and sf6 ? Thanks so much!!