tapparelj/gr-lora_sdr #68: LoRa spreading factor detector

alphapats, 2023-12-12:
I want to create the gnuradio block which can automatically detect the spreading factor of detected LoRa burst signal, and then based on spreading factor decode the signal?It can be completely brute force approach where it automatically tries all SF one by one and match the header to identify correct SF.
Can you assist me with files which I can modify to achieve atleast first part ie detection of SF (5 to 12) and in next phase, I can use your current code to decode the burst with detected SF?

tapparelj, 2023-12-13:
Hello @alphapats,
I see a very simple way to do it without having to write much code (but obviously that wont be the most efficient way).
You can create a flowgraph with 5 chains of receivers. Each will manage one spreading factor.
Between the USRP_source (or any other SDR sample source) each of the 5 chains, you add a block that would to the following:
- Try to demodulate LoRa symbols (assuming one SF) 
- Buffer at least the length of the preamble upchirps in the block.
- If you see multiple repetitions of the same symbol value (probably caused by the upchirps repetition in the frame preamble) you assume that a frame is present with the SF you tried
- You then just output the buffered samples and any following samples for some time (maybe the maximum length of a frame).

With a setup like that, only the receiver chain of the potentially found SF will run and the others will just stay Idle. 
I know that this is not the optimal way to do it but only a simple implementation that came to my mind. It wouldn't require to touch anything in the currently existing blocks, but only for you to write a simple additional block.

I hope this can give you a starting idea of what you could do to avoid having to deal with anything complicated.

alphapats, 2023-12-14:
Thanks @tapparelj . will try it out and get back in case of any issues

tapparelj, 2024-03-07:
Hello @alphapats,
I will close this issue for now, given that there hasn't been any update. 
But don't hesitate to reopen it if needed!

closed 2024-03-07 by tapparelj