tapparelj/gr-lora_sdr #70: Real Time Recording of Live LoRa Signals

jds0987, 2023-12-28:
Hi,

I have been working around with your code and its well written. Kudos to the efforts put in by you. 

However, I am raising this issue as I have been trying to record live LoRa transmissions irrespective of the spreading factor (SF 5 to SF 12) but i have not been able to achieve a breakthrough in this. 

Can you modify the code to help me achieve the same?

Regards

tapparelj, 2024-01-08:
Hi,

From what I understand you could do the following:
- All spreading factors should get recorded simultaneously? If yes, you would need to use 8 _frame_sync_ blocks to detect the the incoming signals, one for each SF.
- If you only want to record incoming signals, the easiest would be to 
  - modify lib/frame_sync_impl.cc to set the variable _m_received_head_ to true, so you don't need to wait for the header to be decoded 
  - set the variable _m_symb_numb_ to the maximum number of symbol a frame can contain, which can be computed by :
  m_symb_numb = 8 + ceil((double)(2 * m_pay_len - m_sf + 2 + !m_impl_head * 5 + m_has_crc * 4) / (m_sf - 2 * m_ldro)) * (4 + m_cr);
-If you want to save the incoming samples without any correction, you can change the _DETECT_ state to directly jump to the _SFO_COMPENSATION_ state and obviously remove any SFO compensation.
- Connect each of the _frame_sync_ block to a _file_sink_ for saving the samples.

I hope this gives you some idea on how you could achieve the recording of frames. 
Let me know if this is not what you meant by recording of live signal or if you encounter any issues.

jds0987, 2024-01-08:
Hi....

Thanks for the reply.

Should i try implementing these changes in master branch or the SF5 and SF6 Branch.

If i go ahead with the master branch then, will it able to record for SF5 and SF6 as well?

jds0987, 2024-01-08:
Also 

Let me know how can the name of the recordings of file sink be stamped with the following content

i.e.

Time
Date
Spreading Factor
Centre Frequency?

tapparelj, 2024-01-08:
You can use the master branch, only the decoding of sf5/6 was not working, but the preamble detection was fine.
You can also use the sf5/6 branch if it's more convenient for you.
For the naming, you can either use [Tagged_File_Sink](https://wiki.gnuradio.org/index.php/Tagged_File_Sink) and use different folders for the SF and Fc. Else you can simply write the samples to a file in the frame_sink_impl.cc with the name you want and connect the output to a _null_sink_.
In addition, you probably want to increase the number of consecutive upchirps required for detection (_m_n_up_req_ l.45) to avoid too many false triggers, especially for smaller SFs.

jds0987, 2024-01-08:
Hi 

I have been able to record the file using a file sink after frame sync as u had explained.

But when I am using this file as file source in the lora_RX.grc, I am not able to retrieve the decoded output.

tapparelj, 2024-01-08:
By default the preamble is not output, you have to manually output it by copying to the output buffer and return the value.
After thinking about it, as the operation you want to do is very simple, it would probably be easier to write a simpler block than _frame_sink_ rather than modifying it. You can use it as a template were you can take a few of it's functions and do only the following in the work function:
- Keep a buffer of ~8 symbols
- Dechirp and compute the DFT of each new symbol (you can use the function get_symbol_val() to do that)
- If you see ~6 repetition of the same value +- 1
  - output the buffered samples
  - output the following input samples until you have the maximum frame size.
  
Doing just the following steps would probably be easier to be certain that you save exactly what you want.

jds0987, 2024-01-27:
Hi @tapparelj 

When you say about writing a new block, do you mean to say replacing the existing Frame sync block with a new one in the existing branch?

If yes, then can you guide me step by step how to go about it?

I will just again explain my requirements again for your comprehensive understanding

1. Requirement of recording LoRa Signals(only SF5 for now) in a band of 20 MHz using good SDR and antenna

2. Creation/ using an existing flowgraph with input as a file source to decode these recorded LoRa signals.

Given these two requirements, can the existing code be modified for fulfilling these requirements as writing a new code will involve more complexities and will be time consuming 🕰️.

alphapats, 2024-01-29:
> By default the preamble is not output, you have to manually output it by copying to the output buffer and return the value. After thinking about it, as the operation you want to do is very simple, [...]

Can you elaborate on this? Writing samples to file once LoRa chirp is identified?

tapparelj, 2024-01-30:
@jds0987 
for both 1. and 2. you will need to have one rx chain per channel (carrier frequency). For storing the data over multiple channels, you can as an example, take your signal samples at a high rate 20MHz and for each center frequency, shift the signal by multiplying it with a complex exponential, decimate it and low-pass filter it and feed it to the frame sync.
1. If you don't want to create a new block, the minimum modifications required to do to the frame_sync block would be: 
    - Create a buffer of at least the size of the preamble (12.25 symbols) where you store the most recent samples.
    - Instead of going from the DETECT to SYNC state, you go into a new state that you create and would just copy what is in your buffer to the output for some time (based on the maximum duration of a frame).
    - Connect the output of the block to a filesink.
2. You just need one rx chain for each channel you stored in files.
If you don't want to store the samples of each channel individually, I don't see an easy way to do it without creating your own block.

tapparelj, 2024-01-30:
@alphapats 
I'm not sure to get which part is not clear, can you be more specific? 
To write samples to a file you can either do it directly inside the block (as you would normally write into a file in C++)  or copy the samples to the block output and then connect that output to a filesink block.

alphapats, 2024-01-30:
> @alphapats I'm not sure to get which part is not clear, can you be more specific? To write samples to a file you can either do it directly inside the block (as you would normally write into a file in C++) or copy the samples to the block output and then connect that output to a filesink block.

I have tried to modify frame_sync block to write samples directly to file once LoRa samples are detected. Code is getting compiled and is creating new files when LoRa burst is detected but when I am trying to read those files, code is not able to process those files and is giving error.
I have added these additional lines in DETECT case of m_state switch. Can to 
`if (symbol_cnt == (int)(m_n_up_req))
                {
                    additional_upchirps = 0;
                    m_state = SYNC;
                    symbol_cnt = 0;
                    cfo_frac_sto_frac_est = false;
                    k_hat = most_frequent(&preamb_up_vals[0], preamb_up_vals.size());
                    memcpy(&net_id_samp[0], &in[int(0.75 * m_samples_per_symbol - k_hat * m_os_factor)], sizeof(gr_complex) * 0.25 * m_samples_per_symbol);

                    // perform the coarse synchronization
                    items_to_consume = m_os_factor * ((int)(m_number_of_bins - k_hat));
                    ppFile = fopen (fname.c_str(),"ab"); 
                   //fwrite(&in_down[0], sizeof(gr_complex), m_number_of_bins * sizeof(gr_complex) , ppFile);     
                    fwrite(&in[0], sizeof(gr_complex), m_symb_numb * sizeof(gr_complex) , ppFile);     //write input samples to file fname
		    fclose(ppFile);
`

Can you highlight what mistake I am doing here?

tapparelj, 2024-02-13:
Hello @alphapats,
by reading the definition of fwrite, it takes the number of element to write as third argument. 
Here you try to write in the file m_symb_numb * sizeof(gr_complex) which will write only 8*m_symbol_numb samples to your file.

If you want to write the complex samples of m_symbol_numb, you should use m_symbol_numb*m_samples_per_symbol.
However, all sample may not be available in the input buffer in* when you try to write the file. You should check how many symbols are available in the input buffer, checking the value ninput_items[0] and write only the ones available. You can then keep track of how many you wrote and, once you wrote m_symbol_numb, stop copying sample and switch back to the DETECT case.

tapparelj, 2024-05-16:
I'll close this issue given that there has been no follow up since a long time and as it was more a help request that an issue.

closed 2024-05-16 by tapparelj

jasemanoah, 2026-03-05:
Hey I could make some changes in the code in order to record the entire LoRa signal along with the preamble as you mentioned some steps earlier, Could reuse the same signal to decode the data. But there's something which isn't showing me right in the inspectrum unless properly adjusted. It shows opposite symbols as reflection around the actual signal. I've uploaded both of the images fyi. 

[image]
[image]