tapparelj/gr-lora_sdr #71: Header Generation Process

jds0987, 2023-12-28:
Hi @tapparelj 

Can u pls confirm 

Does the header remain same for a specific no of bytes in the payload irrespective of the content of the payload.

That is to say

If payload is abcd or wxyz, will the header remain same as the no of bytes of payload is the same but the content is different?

tapparelj, 2024-01-08:
Hi,

yes both payload would give the same header.
The explicit header only contains: the payload length (in bytes), if there is a CRC for the payload, the coding rate used and a checksum of the **header**.

tapparelj, 2024-01-18:
I'll close this issue as this was quite a specific question. Don't hesitate to reopen it if the answer is not clear.

closed 2024-01-18 by tapparelj