tapparelj/gr-lora_sdr #73: When the tx_gain of the USRP B210 is relatively low, the communication distance between the USRP (as the transmitter) and the SX1280 is particularly short.

linhl1218, 2024-01-08:
We have two SX1280 RF boards, with the transmit power of the SX1280 set to 0dBm. The signal-to-noise ratio and power for communication between the two boards are both quite normal. 

For instance, when the distance between the two SX1280s is less than 2cm, the output result is as follows.
```bash
2024-01-08 16:31:29,801 # ==============================================================================
2024-01-08 16:31:29,812 # Received:" 0x00 0x00 0x00 0x00 0x00 0x00 0x00"(8 bytes) - [RSSI: -30, SNR: 2]
2024-01-08 16:31:29,830 # ==============================================================================
```
For instance, when the distance between the two SX1280s is 1m, the output result is as follows.
```bash
2024-01-08 16:37:40,759 # ==============================================================================
2024-01-08 16:37:40,769 # Received:" 0x00 0x00 0x00 0x00 0x00 0x00 0x00"(8 bytes) - [RSSI: -60, SNR: 2]
2024-01-08 16:37:40,788 # ==============================================================================
```
However, when the USRP B210 is used to send LoRa packets to the SX1280, the signal-to-noise ratio and power of the packets received by the SX1280 are very low. They can barely communicate when they are very close (<2cm). When the tx_gain is very low, or when the distance between the two is slightly increased (>10cm), they can no longer communicate.

For instance, when the distance between the SX1280(recevier) and the USRP(Transmitter) is less than 2cm, the output result is as follows.
```bash
2024-01-08 16:40:12,844 # ==============================================================================
2024-01-08 16:41:44,720 # Received:" 0xff 0xfe 0xfc 0xf8 0xf0 0xe1 0xc2"(8 bytes) - [RSSI: -91, SNR: -12]
2024-01-08 16:40:12,871 # ==============================================================================
```

Why and how to solve this problem？

tapparelj, 2024-01-08:
Hello @linhl1218,
I've never used one of the sx1280, but based on the datasheet of the chip
>p. 135
GetPacketStatus()
The command returns the following parameters:
snrPkt Estimation of SNR on last packet received. In two’s compliment format multiplied by 4.
Actual SNR in dB =snrPkt/4, noting that only negative values should be used and rounded to the nearest integer.

I'm not sure how you get the 2dB SNR as they mention that they should only be negative values. 
Then I would guess that the 2dB is just the maximum value reported (did you see more than 2dB sometimes?)

Then for the B210, I would verify that the configuration of the output power is also corresponding to 0dBm. Check what gain you configure and what it should correspond to based on the b210 datasheets. Do you use the same antenna for the chip and the b210? If not, that could make a difference.
The output gain should not be caused by the lora_sdr OOT module (the samples sent to the radio should have a magnitude of 1.0), unless you added some scaling after the _modulate_ block.

I hope this gives you some hint on what to look for.

linhl1218, 2024-01-09:
Hello @tapparelj,
Thank you for your help! We subsequently found that when the gain of the USRP TX is very small, the signal amplitude received by the receiver is also very small. When the gain reaches 68dB, the signal amplitude emitted by the USRP is similar to that of the SX1280. The specific situation can be referred to the following two figures. 
[image]

[image]
When the gain of the USRP TX is 50dB, the average signal amplitude detected by the receiver is 0.00014945713, while the amplitude of the SX1280 is 0.0006099344. When the USRP TX gain is 68dB, the average signal amplitude detected by the receiver is 0.0005628494.
Therefore, we suspect that the problem is not with the software, but with the hardware.

tapparelj, 2024-01-09:
From what I can see in the datasheets of the B210, the max output gain is only mentioned as ">10dBm" and the gain range can be given in the range  [0,....,89.8] dB. Therefore, it's not surprising that you have to configure a high gain value as when you use a gain of 68dB you can expect that the output power will be greater than (68-89.8)+10 = -11.8dBm.

linhl1218, 2024-01-09:
So that’s how it is! I greatly appreciate your help!!!

tapparelj, 2024-01-18:
I'll close the issue, but don't hesitate to reopen it if you encounter other problems.

closed 2024-01-18 by tapparelj

SpongeBobaa, 2024-09-25:
I am a novice trying to learn Lora network technology. I have found some information that Lora can support transmission for tens of kilometers. However, based on the working distance mentioned in this question and my own use of USRP for transmission, the transmission distance is only a few meters. Why is this?

[image]

tapparelj, 2024-09-25:
Hello @SpongeBobaa,

If you only have a range a few meters, there is a problem with your configuration.
When using USRPs 2920 (or N210), i could reach an outdoor range of ~1.2km (non-line-of-sight) without much difficulties with SF7 (without trying to really find the limit) using simple omnidirectional antennas and 14 dBm transmit power. 
Therefore, something seems off if you only reach a few meters. You might want to double check the antenna port, the antenna frequency range,  Tx gain, etc...