tapparelj/gr-lora_sdr #75: getting nothing terminal window after running setup

Piyushchauhan1999, 2024-01-18:
Setup Done
[image]
Output Getting

[image]

Piyushchauhan1999, 2024-01-18:
physical setup
![WhatsApp Image 2024-01-18 at 5 32 41 P
signal received
[image]
M](https://github.com/tapparelj/gr-lora_sdr/assets/83907036/43515708-d898-44ba-88aa-977b7d56a901)

Piyushchauhan1999, 2024-01-18:
signal Revived
[image]

tapparelj, 2024-01-18:
Hi @Piyushchauhan1999,
first question to see if the installation went correctly: is the flowgraph tx_rx_functionnality_check.py working?

Piyushchauhan1999, 2024-01-18:
[image]
I think its fine

tapparelj, 2024-01-18:
Good, then it's easy to have some mismatch of configuration between the commercial transmitter and the receiver. 
Here are a come things to double check:
- the Tx center freq and bandwidth (seems ok from your plot)
- the sync word (sometime called network identifiers): here the receiver expect 0x12, that value needs to match what the transmitter is set to use.
I noticed that you use a quite high sampling rate. I've never used such a large oversampling factor so I would try using a sampling rate much smaller to avoid picking up a lot of out-of-band noise (e.g. 1MHz -> os_factor = 4).

Some hex values are printed in the terminal, you changed the code to print them or you use a different branch than the master?

If you have a spare hackrf, using them for transmitting and receiving may help to identify if the issue comes from the LoRa chip config or the gnuradio one.

tapparelj, 2024-01-18:
this is very strange. 
Can you try to see if the transmitted frame seems strange? 
As example on my side using cubicSDR (easy to install with just apt install) if I set the SF to 12 and use a simple setup with two LimeSDR Mini, I'm able to clearly see the transmitted chirps as:
[image]

Piyushchauhan1999, 2024-01-19:
with sf 7 i am not getting any timeout 
[image]
but with sf12 i am getting  timeout
[image]

tapparelj, 2024-01-19:
I've never used a Hack RF One but it seems that they don't recommend to use a sampling rate lower than 8MHz (based on [this](https://hackrf.readthedocs.io/en/latest/sampling_rate.html)), but to decimate and filter after the soapy source in gnuradio.

Are the timeout showing up at regular time interval? If the time interval corresponds to the period between frames, this is caused by the flowgraph not generating constantly samples, but only for a frame (there is a tag present in the stream that fixes this "issue" for the UHD USRP sink, but I don't see a similar option for the soapy hackrf sink)

Even with the error, are you able to get a trace of what you are transmitting?

Piyushchauhan1999, 2024-01-19:
this is a real lora transmission from a lora device that my blade rf is able capture
[image]
and i am getting this error in No gui mode

[image]

Piyushchauhan1999, 2024-01-19:
can you send me the details of the sdr that you are using its name model and settings you are using with its specific firmware and if you can provide link or provider from where or whom you have purchased it

tapparelj, 2024-01-19:
The error you get is caused by having a buffer too small between the Soapy source and the frame sync. You can fix it by setting the _minoutbuf_ parameter of the soapy source (under the Advanced tab) to something like : int(np.ceil(samp_rate/bw*(2**sf+2))).

As SDR receivers, I've used:
- NI USRP-2920
- USRP B200mini
- LimeSDR USB
- LimeSDR Mini
For the RF settings, I mostly use 125kHz bandwidth, 500kHz sampling rate and a carrier frequency of 868.1MHz

Piyushchauhan1999, 2024-01-24:
Hey tapparelj,
I was looking to buy LimeSDR Mini. But Lime SDR USB and LimeSDR Mini have been discontinued. The RF parameter for LimeSDR Mini 2.0 is same as LimeSDR Mini. Have you tested LimeSDR Mini 2.0?

tapparelj, 2024-01-24:
I don't have a LimeSDR Mini 2.0 unfortunately, but they still use the same RF chip LMS7002M so they should not behave differently.

closed 2024-03-04 by tapparelj

Piyushchauhan1999, 2024-03-06:
Hello,
I have purchased a LimeSDR mini 2.0 but still I'm not able to get the results 
[image]

Piyushchauhan1999, 2024-03-06:
When using LoRa Rx block I was able to get the result 
[image]
But If if keep the Implicit header to NO keeping the payload length and coding rate same I was not able to get the results
[image]

Piyushchauhan1999, 2024-03-06:
also the LoRa Rx block with implicit Header = NO is not working for sf 5 and sf 6