tapparelj/gr-lora_sdr #79: HackRF CRC issues

radio-satellites, 2024-02-10:
Hi!

Thanks for the wonderful work!

I'm wondering about why my HackRF receiver nearly always produces wrong CRCs (sometimes correct ones, though), but my RTL SDR receives the same signal just fine. 

I initially thought that the DC imbalance was the issue on the HackRF, so I tried receiving the same signal off to the side and centering it in GNURadio. However, this provided no measurable effect. 

Another interesting observation I made is that this seems to affect only the long packets, the shorter ones get through with zero packet loss. 

Interested in hearing your ideas on this!

Best,

Sasha

tapparelj, 2024-02-13:
Hi @radio-satellites,
Since you mention that only long packets suffer, it makes me think of an issue related to sampling frequency offset (SFO) as the more time passes, the more error you accumulate. In the current implementation, I assume that both the carrier frequency and the sampling frequency are generated using the same reference oscillator. This is used to get an estimate of the SFO based on the carrier frequency offset (CFO) estimated in the preamble. 
Maybe this assumption is not correct for the Hack-RF?
If you add a _Tag Debug_ after the _Frame_sync_ block, it should print the estimated CFO what value does it give?

radio-satellites, 2024-02-16:
Hey @tapparelj !

Thanks for your reply!

I'm not quite sure whether or not a different reference oscillator is used in the HackRF. 

So far I am using the integrated LoRa Rx block. Let me assemble the entire GnuRadio flow with those blocks and see what happens. 

Thanks!

Sasha

tapparelj, 2024-05-16:
Hello, I'll close this issue given that there has been no update since some time.
Don't hesitate to reopen it if needed.

closed 2024-05-16 by tapparelj