tapparelj/gr-lora_sdr #80: Erroneous Decoding for LDRO Enabled Transmissions

Pkindian, 2024-02-20:
Hello @tapparelj 

I am testing your software decoder for LDRO enabled Transmissions and found out that the output is erroneous and garbled. I have also attached the recorded transmission for your testing to fix LDRO issues.

**Transmitted msg in ASCII (hex) at SF 5 with LDRO enabled.**
  **"HELLO WORLD"**

**Msg Received in HEX in LDRO Enabled SX1268 hardware receiver.** 
**48 45 4c 4c 4f 20 57 4f 52 4c 44** // corresponds to HELLO WORLD, that is the transmitted message.

**Erroneous output received using feature/sf5_6_sx126x.** 
rx Hex: 2a 85 c9 66 07 88 28 d0 2e 9d 6d d2 f3 e3 81 88 58 1c f5 eb 13 6d 86 9f 80 95 3a de e0 bc b3 0e 75 64 4b 2b 70 8e eb 41 8e fe 2f 92 12 f0 8e 1f 7d d7 e8 25 8d db a6 28 85 44 6b c5 66 9c 8c 3e 74 7b ed 9b cf 36 1d d6 07 e4 4d fa dd 7e 56 4c bd ba f2 4b 6d ce 7c 19 1f 34 da 38 62 1c 7e 95 d6.

Recording has been done using RTL SDR with following parameters

**Samp Rate.**      2 MHz
**Centre Freq**     450 MHz
**Bandwidth**        125 KHz

Probably the LDRO would require more fixes in this branch of your code.

tapparelj, 2024-03-04:
Hello @PKtheanonymousdeveloper,
this problem with LDRO and sf5-6 that has also been reported in this issue #76.
It's in my todo list and I will have a look when I find some time.
Did you, by any chance, tries to use the LDRO with SF>6? If yes, is the output correct or also wrongly decoded?

tapparelj, 2024-10-04:
Hello,
as it is a duplicate issue and there has been no feedback for some times I will close it.

closed 2024-10-04 by tapparelj