tapparelj/gr-lora_sdr #83: ModuleNotFoundError even after loading PYTHONPATH

sourya1712, 2024-02-28:
I followed all the steps mentioned in your repo and did an install of the gr-lora_sdr module. The installation path is:

"Installing: /usr/local/lib/cmake/gnuradio-lora_sdr/gnuradio-lora_sdrConfig.cmake"

I did a sudo make install.

The modules were not appearing in GNU Radio companion. So, I included the lines of code with the tag [grc] and now they appear in the Companion. However, whenever I run a code with the Lora modules, I get the following error:

Executing: /home/jammy/miniconda3/envs/gr310/bin/python -u /home/jammy/Desktop/loratest.py

Traceback (most recent call last):
  File "/home/jammy/Desktop/loratest.py", line 34, in <module>
    import gnuradio.lora_sdr as lora_sdr
ModuleNotFoundError: No module named 'gnuradio.lora_sdr'

I am using Ubuntu 22.04.

tapparelj, 2024-03-04:
Hello @sourya1712,
from what I see, you would like to use a conda environment, is it correct?
If yes, when you ran the `cmake ..` did you include the conda prefix path? As : 
```
cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX
```
When you run `make install` what is the installation path mentioned (the path before _libgnuradio-lora_sdr.so_
. You can use `make install | grep libgnuradio-lora_sdr.so
` to easily see it)? (btw, if you use a conda prefix in your home, you don't need to run as sudo)

sourya1712, 2024-03-04:
> Hello @sourya1712, from what I see, you would like to use a conda environment, is it correct? If yes, when you ran the `cmake ..` did you include the conda prefix path? As :
>
> ```
> cmake .. [...]

Thank you, your instructions solved the issue.

closed 2024-03-04 by sourya1712