tapparelj/gr-lora_sdr #84: Chirps not appearing in Waterfall

sourya1712, 2024-03-02:
Hi, I am trying to view the LoRa Tx chirps using a Waterfall sink in GNU Radio. But the chirps do not appear in the sink. Do you have any idea how I could probably see the chirps? I am using GNU Radio on WIndows.

tapparelj, 2024-03-04:
Hello,
I've never used gnuradio on windows so I only suppose you have access to similar blocks as on linux.
Gnuradio _QT GUI Waterfall Sink_ is not ideal to visualize temporary signals, but you should be able to see easily the chirps for the largest SFs (e.g. sf12). If you take the flowgraph from tx_rx_simulation.grc, set the _File Source_ Repeat to Yes, connect a _QT GUI Waterfall Sink_ after the Throttle, and set its Update Period to 1e-6 you should see something similar to this (you can change the Samp Rate of the _Throttle_ to make it move slower):

[image]

sourya1712, 2024-03-04:
Also, why does the following flowgraph not print anything when the SF is increased above 9?
[image]

Uptil an SF of 9, it prints the data, the header and checksum valid. above 9, it stops printing anything.

tapparelj, 2024-03-05:
when you use the text file as input, can you try adding a comma at the end of the text? else can you try with the example file located in data/GRC_default/example_tx_source.txt.

For the block of the entire Tx and Rx all SFs works on my side. Can you run the flowgraph in a terminal with SF>9 with `python3 tx_rx_hier_functionality_check.py ` and paste the output here?

sourya1712, 2024-03-06:
> when you use the text file as input, can you try adding a comma at the end of the text? else can you try with the example file located in data/GRC_default/example_tx_source.txt.
>
> For the block of [...]

Hi.
I followed your instructions and test with your example file. It works perfectly with SF 12. Also, I tried adding commas between segments in my own text file. It works after putting a comma. I have attached the screenshots as follows:
[image]
[image]

Could you help me with a few other things? I just want to generate one upchirp or a series of upchirps in the waterfall model. How do i do that? Also what about only downchirps?

sourya1712, 2024-03-06:
> when you use the text file as input, can you try adding a comma at the end of the text? else can you try with the example file located in data/GRC_default/example_tx_source.txt.
>
> For the block of [...]

However the hierarchichal functionality check does not work. It just abrupty ends.

`Generating: '/home/jammy/gr-lora_sdr/examples/tx_rx_hier_functionality_check.py'

Executing: /usr/bin/x-terminal-emulator -e /usr/bin/python3 -u /home/jammy/gr-lora_sdr/examples/tx_rx_hier_functionality_check.py

>>> Done (return code 1)`

This is what i get in the terminal.

tapparelj, 2024-03-07:
Hello @sourya1712,
currently, the _modulate_ block read any decimal value at its input and if it is a beginning of a frame, he prepend the preamble required. You could fairly easily modify this block to:
1. Not add the upchirps, network identifers and downchirps before the frame
    - This would allow you to just use a _Vector Source_ with the values you want to generate chirps with.
2. Make that when you input negative values, it generate the normal chirp, but afterward get it's complex conjugate (this would create downchirps). You can easily compute the conjugate of a complex vector using  [this function](https://www.libvolk.org/docs/branch/main/volk_32fc_conjugate_32fc.html)

tapparelj, 2024-03-07:
> However the hierarchichal functionality check does not work. It just abrupty ends.
>
> `Generating: '/home/jammy/gr-lora_sdr/examples/tx_rx_hier_functionality_check.py'`
>
> `Executing: [...]

You probably get that in the terminal window integrated in gnuradio-companion. Can you instead:
- open a terminal
- `cd /home/jammy/gr-lora_sdr/examples/`
- `python3 tx_rx_hier_functionality_check.py`

This should print the reason of the crash

closed 2024-03-18 by sourya1712