tapparelj/gr-lora_sdr #85: RE:getting nothing terminal window after running setup

Piyushchauhan1999, 2024-03-06:
Hello,
I have purchased a LimeSDR mini 2.0 but still I'm not able to get the results
[image]

When using LoRa Rx block I was able to get the result
[image]
But If if keep the Implicit header to NO keeping the payload length and coding rate same I was not able to get the results
[image]

also the LoRa Rx block with implicit Header = NO is not working for sf 5 and sf 6

Piyushchauhan1999, 2024-03-06:
https://github.com/tapparelj/gr-lora_sdr/issues/75

referencing the closed issue

tapparelj, 2024-03-07:
Hello,
the bandwidth and sampling rate in your flowgraph lora_RX look strange.
- If you send with a standard LoRa transmitter, you should have a LoRa bandwidth of 125, 250, or 500kHz. Here 800kHz is surprizing.
- The value sampling rate of the SDR, the LoRa bandwidth and the os_factor should be related by os_factor = samp_rate/bw, os_factor an integer. I would advise an oversamplinf factor of at least 4 for the synchronization to work well.

For the second question it's probably the same issue as the first flowgraph.
For the last part. SF 5 and 6 are special and depend on the chip you use. e.g. the sx1276 don't support explicit header. the sx1262 does support it, but the implicit header is not compatible between the two generations. So I would need more information on your setup to debug that.

Piyushchauhan1999, 2024-03-12:
Hello,
I am able to receive output of lora_RX block (header and payload) in console. It works great!
But Now I want to store the output data in a variable or a text file. 
Please tell me a way to do this.

The image below shows what I am referring to output data.
[image]

tapparelj, 2024-03-12:
If you know the size of the payload in advance, just plug a file sink in the corresponding output stream to save the payload bytes. There is also an optional output for the result of the CRC check.
However, if you wan to also save the header content, you need to create a new block that:
- reads the output stream and look for the tags that indicate the header content and the crc check result (just plug a _tag debug_ after the _CRC verif_ to see the information contained in the tags).
- save the tags information and payload in any format you like.

Piyushchauhan1999, 2024-03-14:
Hey, I did few changes in your code to store the cout statements into a file. But when I access the file I get "ValueError : I/O operation on closed file"
[image]
[image]
[image]

tapparelj, 2024-03-15:
You get the error when running the python code in parallel to the flowgraph?
If yes, this is probably because there writing and reading the same file at the same time should be avoided but I've never try to do it so I can't give you a solution. You will need to search online for a possible way to do it properly.

closed 2024-03-20 by Piyushchauhan1999