tapparelj/gr-lora_sdr #88: Cannot use a file source to replace USRP source

HaoHan9717, 2024-03-11:
Hi,

I have been working around with your code and its well written. Kudos to the efforts put in by you.

However, I recently ran into a issue: when I am playing a signal file (complex float 32) using a file source instead of USRP source, the module of LoRa RX outputs nothing. It is like the programme is frozen but it is actually running with no outputs.

Can you please tell me how I can fix it?

Best Regards

tapparelj, 2024-03-12:
Hello  @HaoHan9717,
it's difficult without any information about how you generated the signal file you are using.
Can you explain how you got it?

tapparelj, 2024-03-13:
Based on your file, the frame is indeed there.
[image]

Something is bothering me regarding the network identifiers (sync words) what is you configuration, because in your file they look like 0x00 which is not standard and the synchronization will fail if the modulated value is zero as it will be indistinguishable from the preamble upchirps.

Did you specify this value (it is the register 0x39 of your chip)? If not can you set it to the default 0x12?
[image]

HaoHan9717, 2024-03-16:
> Based on your file, the frame is indeed there. [...]

Hi @tapparelj 

Sorry for the delayed reply. I have changed _SyncWord_ in my commercial LoRa nodes and tested your way. It truly can detect and recognize the packets in the binary file. 

However, as shown in the above flowgraph, if I attach a _file sink_ with _LoRa RX_ module to get the decoded results, I still get nothing. From my shallow perspective, shouldn't it output the decoded results of the payload? For now, I can only get the information of **Payload length**, **CRC presence** and **Coding rate**, as demonstrated below.

[image]

HaoHan9717, 2024-03-25:
> Hi, If there is an error in the header, the receiver doesn't know how to decode the packet (when using the explicit header mode). As you use an sx1276, it doesn't support explicit header mode for sf 5 and 6 (it's described in page 27 of the datasheets) ![image](https://private-user-images.githubusercontent.com/66671413/314566108-08a91846-41f4-41a6-9221-416f319c11d5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEzNzQzMjgsIm5iZiI6MTcxMTM3NDAyOCwicGF0aCI6Ii82NjY3MTQxMy8zMTQ1NjYxMDgtMDhhOTE4NDYtNDFmNC00MWE2LTkyMjEtNDE2ZjMxOWMxMWQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI1VDEzNDAyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5OTA5OTI2NmUzMGU5N2EzMGQ1YWNjYjM5YWNkY2Q0OTYzNzNjM2I2MGVhNGY4NWNmYjZmOGExYmY0OThjMWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.L8Ft66u-7KWPewxjtPXEiw-xIhUFIvDuz8ozn1bxzaM) You have to set the _LoRa RX_ block to use "implicit header" and then set the parameter for coding rate, payload length (in bytes) and CRC presence by hand to match the parameters used for the transmission.

Hi,
Sorry for the late reply. I tried to change SF to 7 on my SX1276 node and use the explicit header mode for transmitting. Although the header information of _Payload length, Coding rate and CRC presence_ is correct, I still get nothing for the receiving message, as shown in the below figure.
[image]

I set the payload content of the transmitted packets as **{0x00, 0x01, ..., 0x0F}** on the Tx side, so the receiving message shouldn't be blank. Here is a download link of the signal sample collected from BladeRF: https://drive.google.com/file/d/1IeuMqSGNKMVCsgYVR6MIM7cfIX04Cbw5/view?usp=sharing
All related parameters are: SF7, coding rate 4/5, sampling rate 1M, bandwidth 250k, SyncWord 0x12.

Thanks again for your time on making a perfect open-source codebase of LoRa. 
Hope you can provide me some advice on how to solve this.

HaoHan9717, 2024-03-27:
> Hi, the values printed in the terminal are converted to ASCII and the characters 0x00 to 0x0F are not printable characters. The values will be output by the block and you can save them using a filestream for example. If you use the hierarchical block lora_Rx the option to print the value received as hex values is missing. You can either replace that block by the complete Rx chain (and choose this parameter) or (as a quick hack) you can go inside python/lora_sdr/lora_sdr_lora_rx.py and replace the line 55 `self.lora_sdr_crc_verif_0 = lora_sdr.crc_verif( self.print_payload, False)` by `self.lora_sdr_crc_verif_0 = lora_sdr.crc_verif(2, False)`
> 
> It should force to print the hex values instead of the ascii ones. Then you can run `make install`.
> 
> Let me know if you have any other issues. But since the CRC is valid, the you receive correctly the payload data now.

Hi, @tapparelj 
You were right. It truly was the problem with hex values. After I imported the decoded data into Python, it can be converted and printed in ASCII ones. 

So lastly, I would like to inquire about one more matter: How can I save the decoded data directly into the ASCII ones? This bothered me because now I am using _File Sink_  module in _gnuradio_ to save the decoded results and I cannot read the saved file (garbled characters) unless I converted it in Python or other tools. I wonder if there is an easier way to read the saved file instead of converting it beforehand.

Many thanks for your patience and helpful solutions !

tapparelj, 2024-03-27:
Hi,
glad to hear that you made it work!

The branch sf5_6_sx126x has an additional block and a modified _crc_verif_ block that allows to save a readable form of the received payload to a file (see the pull request #77). You can use that branch instead of the main if you want (If you don't plan to use SF5 or 6, there will be no issue with that branch).

atambi802, 2024-03-27:
@tapparelj I am trying to use the lora block, I am used @HaoHan9717 file, and got the asic out in a file. However when I do it from a bladerf, I can't get any data to be printed. Is there anything specifically I am doing wrong?

[image]

HaoHan9717, 2024-03-28:
> @tapparelj I am trying to use the lora block, I am used @HaoHan9717 file, and got the asic out in a file. However when I do it from a bladerf, I can't get any data to be printed. Is there anything specifically I am doing wrong?
> 
> ![Screenshot from 2024-03-27 13-50-31](https://private-user-images.githubusercontent.com/133379069/317513535-712257ae-18db-4cc5-846a-357751723a78.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE2NDcyMTgsIm5iZiI6MTcxMTY0NjkxOCwicGF0aCI6Ii8xMzMzNzkwNjkvMzE3NTEzNTM1LTcxMjI1N2FlLTE4ZGItNGNjNS04NDZhLTM1Nzc1MTcyM2E3OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyOFQxNzI4MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMzYzOTUzMDQ2MzY5NWRlYTUyZWQxNjQ4MDdhYTMwYmVhZDMwZGM2MjI4MTExNzJlZmRkNjMxMTA5ODMwOTM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.MzRrR_VDdTITtWJaPBlBUq-cVWIp3R8Obnmfgq7fs2s)

Hi @atambi802 

I think that problem might stem from the _Soapy BladeRF Source_ module. But still, I am not sure. You may need to check the out put from the _Waterfall Sink_ first to see if it really received LoRa signals.

And FYI, I am using the _Osmocom Source_ module to receive LoRa signals and it works well for me. If you are sure the related parameters in the flowgraph are correct, then I suggest you try the _Osmocom_ ones. They are better than the _Soapy_ ones.

HaoHan9717, 2024-03-29:
> Hi, glad to hear that you made it work!
>
> The branch sf5_6_sx126x has an additional block and a modified _crc_verif_ block that allows to save a readable form of the received payload to a file (see [...]

Many thanks! @tapparelj

That definitely solved my problems. And BTW, I think your **gr-lora-sdr** repo is the best open-source LoRa platform for SDRs.

Best Regards.

closed 2024-03-29 by HaoHan9717