tapparelj/gr-lora_sdr #89: USRP not working

sourya1712, 2024-03-18:
Hi!
I have an issue that the USRP transmission is not working with the LoRa Tx. I have attached the screenshots of the same. 

This is my flowgraph:
[image]

The output on the terminal is as follows:
`Executing: /home/jammy/miniconda3/envs/gr310/bin/python -u /home/jammy/Desktop/GRC files/lorausrp.py

[INFO] [UHD] linux; GNU C++ version 10.3.0; Boost_107400; UHD_4.2.0.HEAD-release
[INFO] [B200] Detected Device: B200
Traceback (most recent call last):
  File "/home/jammy/Desktop/GRC files/lorausrp.py", line 218, in <module>
    main()
  File "/home/jammy/Desktop/GRC files/lorausrp.py", line 196, in main
    tb = top_block_cls()
  File "/home/jammy/Desktop/GRC files/lorausrp.py", line 91, in __init__
    self.uhd_usrp_sink_0 = uhd.usrp_sink(
RuntimeError: EnvironmentError: IOError: empty input file /home/jammy/miniconda3/envs/gr310

>>> Done (return code 1)`

sourya1712, 2024-03-18:
[image]

tapparelj, 2024-03-19:
First, similarly to your previous issue, (https://github.com/tapparelj/gr-lora_sdr/issues/84#issuecomment-1983193965) please run the python script in a terminal and not directly in grc to see the complete error logs.
Then, 
- are you sure your USRP can use a 70dB gain? 
- can you set the "TSB tag name" to frame_len (as in the example file tx_rx_uspr.grc)?

sourya1712, 2024-03-25:
> First, similarly to your previous issue, ([#84 (comment)](https://github.com/tapparelj/gr-lora_sdr/issues/84#issuecomment-1983193965)) please run the python script in a terminal and not directly in grc to see the complete error logs. Then,
> 
>     * are you sure your USRP can use a 70dB gain?
> 
>     * can you set the "TSB tag name" to frame_len (as in the example file tx_rx_uspr.grc)?

Hi. I ran the python script on the terminal and set the TSB tag name to frame_len. The log is as follows:
[image]

sourya1712, 2024-03-25:
Also, is there any way to install your OOT module without using the conda environment?

tapparelj, 2024-03-26:
Hello,
there seems to have an error with the USRP configuration to narrow down the problem:
- Can you use your USRP with another flowgraph? (not related to the code of this repo)
- Have you tried setting the Sync option of the usrp sink to "no sync"?
- Are you able to see your USRP when using uhd_usrp_probe or uhd_find_device? (those are terminal command lines)

> Also, is there any way to install your OOT module without using the conda environment?

Of course, as long as you have GNU Radio installed and cmake, etc.. (listed in the readme) you can build it as any other OOT module.

tapparelj, 2024-05-16:
I'll close this issue as there has been some time since the last follow-up.

closed 2024-05-16 by tapparelj