tapparelj/gr-lora_sdr #90: can’t recv LLCC68 message

11tools, 2024-04-07:
[]
[image]

I use  hackrf and plutosdr make this. Both. Them can't recv LLCC68 mod

tapparelj, 2024-04-09:
Hello @11tools,

from your screenshot I can't notice anything strange, therefore, I would guess that there is a configuration mismatch between the transmitter and receiver. So here are a few questions:
- Is the tx_rx_functionality_check.grc flowgraph working?
- is the frame visible on the waterfall sink? (using a larger spreading factor may help seeing it)
- what is the sync word used by your module? (cf datasheets)
[image]
- If you save the samples you receive in a file and then plot their instantaneous frequency can you see the chirps (when the transmitter is close to your receiver to get a large SNR)?

11tools, 2024-04-12:
Hello @tapparelj .

  Thanks . I changed the Spreading factor to 8,9,10,11,12... still failed no message. maybe the sync word is difference

  I had made a flow with Lora Tx and a Lora Rx on PlutoSDR. it can recv message what Lora Tx send.

  My real goal is use gr-lora_sdr to recv startlink 150M Lora signal. because 150M lora module is hard to buy. 
   
  have anyone try this?

tapparelj, 2024-10-04:
As it has been some time without follow-up with more information on your setup/problem, I'll close the issue. Don't hesitate to reopen it if needed.

closed 2024-10-04 by tapparelj