tapparelj/gr-lora_sdr #91: Difficulties decoding meshtastic

miweber67, 2024-04-28:
First, thanks for the awesome work!

I've been trying to decode a local meshtastic network and having poor results (<5% of received frames have valid CRCs, even with SNR's exceeding +10dB). Headers always seem to have valid CRCs. I've tried an airspy mini and an rtlsdr on the front end with similar results. I've tried Fs between 1e6 and 6e6 and filtering/decimation to achieve os_factors from 4 to 24. The beginnings of the frames 'look' correct, e.g. the first four bytes are 0xff, which corresponds to a 'broadcast' destination address.

Local meshtastic nodes are using the 'long_moderate' preset, which corresponds to 125kHz with SF=11 and 4/8 coding. The preamble length is claimed to be 16, but manually counting the sweeps in gr_fosphor reveals somewhere between 16 and 18 upsweeps for the preamble. The network ID is '0x2B.' Payload lengths vary between 40 and 70 bytes, typically.

What is a good target number for the frame_sync's 'os_factor?'
Is soft decoding expected to work better than not using it?

I'm not sure if I have a sub-optimal receiver configuration or if there is just some difficulty decoding the particular parameters meshtastic is using. Thanks for any thoughts you have. I can supply IQ recordings if that would help.

tapparelj, 2024-04-29:
Hello @miweber67,
there definitively is something off if you only receive a few correct frames at such a high SNR.

- For the _os_factor_: a oversampling of 4 should be enough to have good performance without too much overhead.
- Yes, the soft-decision decoding should work better (or at least not worse) than hard-decision decoding. And for large receive SNR, it will automatically switch to hard decision to avoid numerical issues.
- 
Some questions:
- Can you share the flowgraph you use, so I can have a look if anything strange pops out?
- Which carrier frequency are you using?
- Also, can you connect a _Tag Debug_ to the output of the _Frame Sync_ and see what value has the CFO estimate (cfo_int and cfo_frac)?

miweber67, 2024-04-29:
@tapparelj , thanks for the response! I can certainly do what you asked.

Please find attached the tag_debug log, console output from the session. 53 frames detected, 5 crc valid. CFO int mostly between -60 and 30. CFO frac varies. 

Carrier frequency is 903.0625MHz. You'll see some frequency shifting blocks in the script, but these are only to support offset collection and channelization for inspecting different channels. The lora decode performance is the same without them. There's also a filter to get rid of adjacent channel interference. Removing the filter drops detections to 33 and valid CRCs to 4.

Also attached is the grc file, although I had to rename it .txt because github wouldn't accept it as .grc.

Thanks much for your help!

[tag_debug.txt](https://github.com/tapparelj/gr-lora_sdr/files/15155948/tag_debug.txt)
[lora-rx-file.txt](https://github.com/tapparelj/gr-lora_sdr/files/15155953/lora-rx-file.txt)

mskvortsov, 2024-05-02:
Sorry for interfering, but I think it's not worth creating a separate issue since my observation might be related to this one.

I have two signals, both contain some meshtastic packets with the same channel settings corresponding to the LongFast preset: SF 11, BW 250KHz, preamble length 16, sync word 0x2b. Center frequency is 869.075 MHz, 1 Msps sampling rate.

[sample2.griq.xz](https://drive.google.com/file/d/1oCERevP0KvZfX0JkUH0haShKYQ9VRmYb) is decoded successfully with the sync word set to expected 0x2b (or [16, 88] in the full form). However, to decode [sample1.griq.xz](https://drive.google.com/file/d/1fiF81PkimhPrtBEaAf9OWR7OirhGl8c_) I need to set the sync word to [16, 2008].

Is it expected? Am I missing something?

tapparelj, 2024-05-03:
Hello @miweber67,
First to answer some questions:
- Regarding the filter: it is necessary to have a a input signal filtered at the LoRa bandwidth to remove noise outside the band of interest (depending which SDR you use, you can configure directly their filter to be smaller than their sampling rate, else adding a low-pass filter before the _frame_sink_ is also fine).
- From the spectrogram, I htink there are 16 upchirps (not 18) and then two modulated upchirps with values [16,88] (for 0x2b) and then the 2.25 downchirps.

I've checked connecting the transmitter to the receiver you configured and it seems to work as expected even at low SNR and with reference clock frequency offset. So I don't see anything strange. Would it be possible to have access to the sample you use that lead to a wrong decoding as well as the payload that is transmitted? (one or two frame would be enough) so that I can identify which part of the receiver seems to fail?

miweber67, 2024-05-03:
@tapparelj , thanks for your efforts! Here are two slices of the capture file with one bad packet each. These slices have been filtered and decimated by 3 to 500kSps, so you should be able to feed them straight in to frame_sync with os_factor 4.

[903062500-0.5MSs-extract-filt-decim-1.c64.gz](https://github.com/tapparelj/gr-lora_sdr/files/15202326/903062500-0.5MSs-extract-filt-decim-1.c64.gz)
[903062500-0.5MSs-extract-filt-decim-2.c64.gz](https://github.com/tapparelj/gr-lora_sdr/files/15202354/903062500-0.5MSs-extract-filt-decim-2.c64.gz)

chrisdimoff, 2025-01-11:
@miweber67 were you ever able to figure this out?

kdmutriy, 2025-02-05:
I have an IQ listing with a filtered packet with Lohr modulation. I have an sx1272 modem. but i don't see any data after module "Frame synk"

[image]
and my signal

[image]

Help me. what am i doing wrong?

tapparelj, 2025-02-14:
@kdmutriy I see a multiple things that make the reception fail.
- Based on the spectrogram, it seems that you use only 6 upchirps for the preamble (instead of default 8). Did you set the _Preamble_len_ parameter of the _Frame Sync_ block to 6?
- The sampling rate at the input of the _Frame Sync_ should be an integer multiple of the bandwidth, such that os_factor*bw = samp_rate. If your input signal is sampled at 994.065 kHz (are you sure about this value?), you have to resample it to 1MHz, for example, using a _Rational Resampler_ block.
- Your _center_freq_ should be the center frequency of the RF signal when it was recorded, probably in a few hundred of MHz.

kdmutriy, 2025-02-17:
@tapparelj 
1. I have a TBS Crossfire transmitter, I recorded its IQ and analyzed it. he is as I showed in the picture.
2. I decided to try to figure out a simpler signal and analyzed the E32 ebyte transmitter.

my results with E32 ebyte:
signal parameters:

bw=500k (I see on the spectrum)
sf=11

- With the help of SDRAngel, I managed to get stable demodulation, but there are problems with decoding. 
- with the help of GNURadio, I get only the inscription and then the processing stops

[image]

My grc:

[image]

H-a-r-sha, 2025-06-21:
Hi @kdmutriy,

I’m facing a similar issue — I was able to properly demodulate the signal and identify both the preamble and sync words. However, I’m running into problems with the decoding part. I’m using an SX1276 LoRa chip and my SDR to capture the packets, but I keep seeing **"Header Checksum Invalid"** on the terminal.

Did you manage to resolve this issue? I’d really appreciate it if you could help me out with this.

Thank you!