tapparelj/gr-lora_sdr #95: Trouble receiving real-time signals

leoposc, 2024-05-14:
Hey, first of all I want to thank for all the amazing work. Your repo is amazing and I am thrilled to use it. Sadly my poor knowledge of digital signal processing lacks too many fundamentals to understand why I can not receive/ decode any frames. I would really appreciate if somebody could look into this and maybe spot any misconfigurations. 

My configurations are:
-`samp_rate`: 1M (since my hack_rf one does not suppport a sampling rate of 500k. I tried to make use of the Keep 1 in M block, so I do not modify the oversampling rate, but it did not help.)
-`center_freq`: 868.1MHz

I am using a dragino humidity and temperature sensor for my testing purposes. On my network server I could receive the messages. The server states the following configurations which I applied to my flowchart:
-`sf`: 12 
-`bandwidth`: 125k

The picture I appended is a waterfall sink - captured while the dragon sensor transmitted its first message to the server. 

[image]

tapparelj, 2024-05-14:
Hello @leoposc,

just to clarify something about this repo, it implement only the physical layer of LoRa, not the MAC layer. If you're not familiar with those concepts, it means if you send a message with a device that uses LoRaWAN and that you can see your message is some web interface, this repo will be able to receive the message, but it will still be encrypted and contains more information than, for example, just a temperature value . This repo is planned to be used with either both transmitter and receiver using it, or using a LoRa device that allows to not use LoRaWAN (the MAC layer of LoRa). By quickly looking, your sensor node seems to be only planned to be used with LoRaWAN. 

Let me know if you were aware of the restriction to the physical layer in that case I would be happy to help you make things work by looking your setup in more details .

leoposc, 2024-05-17:
Hi @tapparelj,

thank you for getting back to me so quickly. I was aware of the fact that the payload data is encrypted and I won't be able to read it without the corresponding key. Since my task is to determine any security risks of the protocol, it is totally fine that I cannot decode the data. 

My understanding is that the preamble and the physical header is not encoded, so I assumed that I would be able to receive the frames with the encoded payload. Isn't that right?

tapparelj, 2024-05-17:
Hello @leoposc,

indeed you should be able to receive frames with some part that is not encrypted (like the node address, and some other info), then some encrypted data that contain the actual payload and some info only addressed to the owner of the transmitting node.

Can you attach a screenshot of your flowgraph such that I can have a look if I notice anything strange ?

leoposc, 2024-05-23:
Hello @tapparelj,

that would be perfect. To be honest, there is not a lot of stuff I changed. I looked into the `tx_info` field of more lora frames received from the network server and could notice that the spreading factor and frequency are not constant. I'll add more frame sync blocks to approach the problem with different SFs (as you described here: https://github.com/tapparelj/gr-lora_sdr/issues/68). Do you know how bad it is when the center_freq does not completely match the actual one? E.g. 867.9 MHz instead of 868.3 MHz?

[image]

tapparelj, 2024-05-23:
Hello @leoposc,
any offset of the center frequency of more than 31.25kHz will prevent the reception of the frame. This margin is usually only planned for imprecision of the reference oscillator of the transmitter and receiver and not an erroneous center frequency.
I don't see any obvious error in your flowgraph but be sure to set the _sync word_ parameter in the _Frame Sync_ block to _0x34_.
You should also set the bandwidth of the _Soapy HackRF Source_ to _bw_ or add a low pass filter with a cutoff of 62.5kHz before the _Frame Sync_ block to remove the out of band noise.

tapparelj, 2024-10-04:
As it has been some time without follow-up, I'll close the issue. Don't hesitate to reopen it if needed.

closed 2024-10-04 by tapparelj