tapparelj/gr-lora_sdr #96: how to get rssi

LeeKoTo, 2024-05-25:
hello,it's a good project, and how can i get rssi

tapparelj, 2024-05-27:
Hello @LeeKoTo,
there isn't an RSSI value computed, but you can get an estimation of the SNR of each frame.
If you uncomment the line 5 of gr_lora_sdr/lib/frame_sync_impl.h "#define PRINT_INFO", and build again, the SNR estimation will get printed in the terminal as:
[frame_sync_impl.cc] 1 CFO estimate: -0.00789291, STO estimate: 116.522 snr est: -5.07804

--------Header--------
Payload length: 14
CRC presence:   1
Coding rate:    0
Header checksum valid!

rx msg: Hello world: 0
CRC valid!

tapparelj, 2024-10-04:
As it has been some time without follow-up, I'll close the issue. Don't hesitate to reopen it if needed.

closed 2024-10-04 by tapparelj