tapparelj/gr-lora_sdr #97: Can't decode a signal

DANTEKSTUDIO, 2024-06-11:
Hi, thanks for this awesome project!
I use RTL SDR V3 Pro and Vega Smart-MC0101 but I can't get any symbols.

tx_rx_functionality_check is working:
[image]

My GNU Radio project:
[image]

My waterfall:
[image]

My LoRa device setup:
[image]
[image]

DANTEKSTUDIO, 2024-06-14:
It works, but I get wrong CRC.

Send packet: 01 63 00 0E 01 02 02 69 CF 6B 66

GNU Radio console:

--------Header--------
Payload length: 24
CRC presence:   1
Coding rate:    1
Header checksum valid!

rx msg: 0x80, 0x2b, 0x8, 0x44, 0x0, 0xc, 0x31, 0x0, 0x52, 0xe4, 0x74, 0xd9, 0x26, 0x2a, 0xf3, 0x7, 0x14, 0x63, 0x84, 0x62, 0xd8, 0x93, 0x1d, 0x37
[31mCRC invalid[0m

tapparelj, 2024-06-16:
Hello,
given that your device uses the sync word 0x34 and that the received payload is longer than what you send, your devices seems to send data using the LoRaWAN MAC layer.
You might want to have a look at [this issue](https://github.com/tapparelj/gr-lora_sdr/issues/95#issuecomment-2110414937) to get more details on what this repo can (and can not do).
Regarding the CRC error, can you try to use a different option for the LDRO  in the _header decoder_ block (if it's on AUTO try disabling it, or set it to AUTO if it was disabled)?

DANTEKSTUDIO, 2024-06-17:
And the last question. I know AppKey and NwkKey. Can I decode bytes of LoRaWAN (the MAC layer of LoRa) using a python block?

tapparelj, 2024-06-17:
Hello,
unfortunately I'm not aware of a python implementation of the MAC frame decoding.
I currently have a version of a transmitter that can send LoRaWAN frames using ABP, however, there is no downlink support yet.
I will publish the code for the transmitter early July. I can keep you updated if it interests you.

DANTEKSTUDIO, 2024-06-17:
[tapparelj](https://github.com/tapparelj), thanks for everything. I'll definitely keep an eye on the progress of the project.

DANTEKSTUDIO, 2024-07-08:
Hello, @tapparelj.
What about the transmitter? When we can test it?

tapparelj, 2024-07-19:
Hello @DANTEKSTUDIO,
sorry for the longer than expected delay. The code for LoRaWAN Tx is available [here](https://github.com/tapparelj/gr-lorawan). There was some change to the phy layer to support the transmission so you will need to update your installation of gr-lora_sdr as well.

For now, the transmissions are limited to the EU bands, but adding support for other bands should not be too hard.
Let me know if you encounter any difficulties or problems.

tapparelj, 2024-10-04:
I'll close the issue as this repo only implements the physical layer of LoRa, so the MAC layer implementation is not an issue of the implementation.

closed 2024-10-04 by tapparelj