tapparelj/gr-lora_sdr #98: InvalidSpec: The package "conda-forge/noarch::adwaita-icon-theme==41.0=ha770c72_1" is not available for the specified platform

Abdull, 2024-06-14:
I'm following the [installation steps as written in `#a303cb395d4139710bf663f5a9b5e1bd59e3feb0`](https://github.com/tapparelj/gr-lora_sdr/blob/a303cb395d4139710bf663f5a9b5e1bd59e3feb0/README.md#installation).

When I run

```
conda env create -f environment.yml
conda activate gr310

# ...

conda install -c tapparelj -c conda-forge gnuradio-lora_sdr
```

... i get the following error message:

```
$ conda install -c tapparelj -c conda-forge gnuradio-lora_sdr
Channels:
 - tapparelj
 - conda-forge
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

InvalidSpec: The package "conda-forge/noarch::adwaita-icon-theme==41.0=ha770c72_1" is not available for the specified platform
```

Same `InvalidSpec` error message appears when executing `conda install -c conda-forge gnuradio
`.

Maybe I'm mixing up steps that should be run *outside* of the `gr310` environment? I don't know if it makes for the `cmake`/`make` commands any difference whether they are executed in a / the `gr310` environment.

tapparelj, 2024-06-16:
Hello @Abdull,
what OS are you using?

Abdull, 2024-06-19:
> Hello @Abdull, what OS are you using?

Debian 12 bookworm, KDE Plasma 5, Wayland.

Via KDE's *System Settings* > *Basic Information: About this System*:

```
Operating System: Debian GNU/Linux 12
KDE Plasma Version: 5.27.5
KDE Frameworks Version: 5.103.0
Qt Version: 5.15.8
Kernel Version: 6.7.12+bpo-rt-amd64 (64-bit)
Graphics Platform: Wayland
Processors: 12 × AMD Ryzen 5 7640U w/ Radeon 760M Graphics
Memory: 27,2 GiB of RAM
Graphics Processor: AMD Radeon Graphics
Manufacturer: Framework
Product Name: Laptop 13 (AMD Ryzen 7040Series)
System Version: A5
```

```
$ uname -a
Linux abdull 6.7.12+bpo-rt-amd64 #1 SMP PREEMPT_RT Debian 6.7.12-1~bpo12+1 (2024-05-06) x86_64 GNU/Linux
```

tapparelj, 2024-06-25:
Sorry for the delay.
I'm not sure to follow what you mean by make/cmake. If you try to install the module using ```conda install -c tapparelj -c conda-forge gnuradio-lora_sdr``` you would not have to build anything yourself.
If you also have problems with the installation of gnuradio using conda I would suggest to try build the module without using conda. 
For that you can follow the installation described in "From Source" and skip the conda installation.
You just have to install gnuradio from a package manager (e.g. sudo apt install gnuradio).
You should then be able to build and install the gr-lora_sdr module.
Let me know if you encounter any issue when using this installation method.

closed 2024-10-04 by tapparelj