tapparelj/gr-lora_sdr #102: Lora hierarchical blocks

atambi802, 2024-07-25:
Hey! 

When I use the lora hierarchical blocks I just don't get any data from it. I am sending lora signals from a rak 2287, and able to listen to in on another rak 2287 without any issues. But when I see it in my bladerf using this, I am not getting any data from the hierarchical blocks. I have tried to see if anything I am doing is vastly different.
This is my flow graph:
[image]

This is what I am sending from the rak 2287 (sx1250):
```Sending 5 LoRa packets on 904100000 Hz (BW 125 kHz, SF 7, CR 1, 0 bytes payload, 8 symbols preamble, explicit header, non-inverted polarity) at 25 dBm```

I am using the newest git pull from a few days ago. And I encountered a seg fault, and some other crazy things compared to the last commit on the master branch.

Thanks so much!

tapparelj, 2024-08-13:
Hello @atambi802,
Sorry for the delay.  When you pulled the new version, did you do a ```make uninstall``` and ```make clean``` before building and installing the new version? If not can you try it?
If this doesn't change anything, can you run the flowgraph tx_rx_hier_functionnality_check (in the examples folder) without any segfault?
Finally would it be possible to run the flowgraph with gdb (```gdb python3``` and then ```run <yourFlowgraph>.py```) and after it crash copy the output of the command ```bt``` ?

tapparelj, 2024-12-04:
As there has been no follow up for some time I will close this issue.
Do not hesitate to reopen it if needed.

closed 2024-12-04 by tapparelj