tapparelj/gr-lora_sdr #103: packages not found

bjyk, 2024-08-16:
Hello, I encountered an issue while downloading version gr-lora-3.8. My Ubuntu version is 18.04 and my GNU version is 3.8. While following the tutorial to run "conda env create - f environment. yml", I couldn't find many packages (even cmake, gnuratio-und, etc.). Is there any solution? Thank you

tapparelj, 2024-08-27:
Hello @bjyk,
If conda fails to create a valid environment, you can try to install packages directly (e.g. gnuradio, cmake, ...). If you install gnuradio, most of the requirement should be installed alongside and I would simply try to build the module and install any package that is reported as missing.
However, the lora_sdr version compatible with 3.8 is quite crude compare to the current version so if you can upgrade to gnuradio 3.10, that would make it much easier to use this module.

tapparelj, 2024-10-29:
As there has been no follow up for some time I'll close this issue. Don't hesitate to reopen it if needed.

closed 2024-10-29 by tapparelj